ARM-software/tf-issues #621: Regarding Trusted Firmware-A v1.6 release

soby-mathew, 2018-08-31:
This is a heads up that we plan to do a release during the first week of October. The aim is to consolidate the work done till now (especially the Spectre workarounds). As part of this, a release tag will be created and release activities will commence from September 7th.  Essentially we will not merge any major enhancements from September 7th till the release is made.

Also in the plan is another release closely following v1.6 which will remove the deprecated interfaces as mentioned in tf-issues #613.

sivadur, 2018-09-04:
@soby-mathew  could you please let me know on the timeline for the release after 1.6 which removes all the deprecated interfaces.

soby-mathew, 2018-09-04:
@sivadur,
We intend to do the release for removing the deprecated interfaces within a week or so following the v1.6. This means that the only change for the new release would be removal of deprecated interfaces with no change in functionality. To minimize disruption to upstream platforms, we might do migration for the trivial cases but don't depend on that. So if you can have a PR towards the end of September for your platform for the migration, that will be great.

Also I have made a comment in tf-issues #613 regarding the MULTI_CONSOLE_API which should reduce the complexity of migration for most platforms.

dp-arm, 2018-09-05:
@sivadur I will be merging your PR today.

antonio-nino-diaz-arm, 2018-09-25:
This PR is the one that removes the deprecated interfaces: https://github.com/ARM-software/arm-trusted-firmware/pull/1587

soby-mathew, 2018-10-02:
The release [v1.6](https://github.com/ARM-software/arm-trusted-firmware/tree/v1.6) and [v2.0](
https://github.com/ARM-software/arm-trusted-firmware/tree/v2.0) are done.

closed 2018-10-02 by soby-mathew