ARM-software/tf-issues #622: While bringing up the secondary cores with ARM IP CCN-508, Linux hangs.

pangupta, 2018-09-04:
Two Observations:
1. When all the RN-F Id are included in the snoop/dvm domain in one go, the Linux works fine.
2. When RN-F mapped to the cluster whose core is coming up, is incrementally included in the snoop/dvm domain, Linux hangs.

Do this issue earlier reported?
Am I placing the call for updating the snoop/dvm domain correctly?

Observation 2 Code level details & Linux hanged call trace:
======================================
Using Function: 
void plat_arm_interconnect_enter_coherency()
{
     ccn_enter_snoop_dvm_domain(1 << MPIDR_AFFLVL1_VAL(read_mpidr_el1()));
}

Linux raises SMC to request runtime secure firmware(BL31) to bring up a new core.

In the BL31, the above function is executed as part of platform hook:
static plat_psci_ops_t _psci_pm_ops = {
...
.pwr_domain_on_finish = _pwr_domain_wakeup
...
};

static void _pwr_domain_wakeup(const psci_power_state_t *target_state)
{
        u_register_t core_mask  = plat_my_core_mask();
        u_register_t core_state = _getCoreState(core_mask);

        switch (core_state) {
        case CORE_PENDING : /* this core is coming out of reset */

                 /* soc per cpu setup */
                soc_init_percpu();

               plat_arm_interconnect_enter_coherency();

                 /* gic per cpu setup */
                plat_ls_gic_pcpu_init();

                 /* set core state in internal data */
                core_state = CORE_RELEASED;
                _setCoreState(core_mask, core_state);
                break;
...
...
}

Linux hangs just after executing plat_arm_interconnect_enter_coherency().

Linux traces:
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.9.62-gfa56a24 (meenakshi@uefi-OptiPlex-790) (gcc version 7.3.0 (GCC) ) #6 SMP PREEMPT Mon Sep 3 14:45:53 IST 2018
[    0.000000] Boot CPU: AArch64 Processor [410fd083]
[    0.000000] earlycon: pl11 at MMIO32 0x00000000021c0000 (options '')
[    0.000000] bootconsole [pl11] enabled
[    0.000000] Memory limited to 2048MB
[    0.000000] efi: Getting EFI parameters from FDT:
[    0.000000] efi: EFI v2.70 by EDK II
[    0.000000] efi:  MEMATTR=0xf037d018 
[    0.000000] cma: Reserved 16 MiB at 0x00000000f9c00000
[    0.000000] NUMA: No NUMA configuration found
[    0.000000] NUMA: Faking a node at [mem 0x0000000000000000-0x00000020841fffff]
[    0.000000] NUMA: Adding memblock [0x80000000 - 0xefc6ffff] on node 0
[    0.000000] NUMA: Adding memblock [0xefc70000 - 0xefcdffff] on node 0
[    0.000000] NUMA: Adding memblock [0xefce0000 - 0xf0313fff] on node 0
[    0.000000] NUMA: Adding memblock [0xf0314000 - 0xf0317fff] on node 0
[    0.000000] NUMA: Adding memblock [0xf0318000 - 0xf033ffff] on node 0
[    0.000000] NUMA: Adding memblock [0xf0340000 - 0xf034ffff] on node 0
[    0.000000] NUMA: Adding memblock [0xf0350000 - 0xf0b2ffff] on node 0
[... 153 lines cut ...]
[    0.195226] ---[ end Kernel panic - not syncing: Attempted to kill the idle task!
[    1.188184] CPU1: failed to come online
[    1.587697] CPU1: failed in unknown state : 0x0
[    1.592273] bad: scheduling from the idle thread!
[    1.597024] CPU: 0 PID: 1 Comm: swapper/0 Tainted: G      D         4.9.62-gfa56a24 #6
[    1.605023] Hardware name: NXP Layerscape LX2160ARDB (DT)
[    1.610474] Call trace:
[    1.612941] [<ffff000008088ce0>] dump_backtrace+0x0/0x238
[    1.618393] [<ffff000008088f2c>] show_stack+0x14/0x20
[    1.623495] [<ffff00000844861c>] dump_stack+0xa0/0xc4
[    1.628596] [<ffff0000080ece8c>] dequeue_task_idle+0x24/0x38
[    1.634312] [<ffff0000080e6408>] do_set_cpus_allowed+0xb8/0x190
[    1.640292] [<ffff000008150b48>] cpuset_cpus_allowed_fallback+0x28/0x38
[    1.646973] [<ffff0000080e66a4>] select_fallback_rq+0x1c4/0x1d0
[    1.652953] [<ffff0000080eb698>] sched_cpu_dying+0x240/0x2d8
[    1.658668] [<ffff0000080bf8b8>] cpuhp_invoke_callback+0x60/0x198
[    1.664823] [<ffff0000080c05c0>] cpuhp_up_callbacks+0xf8/0x130
[    1.670714] [<ffff0000080c143c>] _cpu_up+0xc4/0x130
[    1.675638] [<ffff0000080c1550>] do_cpu_up+0xa8/0xc8
[    1.680650] [<ffff0000080c1580>] cpu_up+0x10/0x18
[    1.685401] [<ffff000009041ef0>] smp_init+0xc4/0xc8
[    1.690326] [<ffff000009030c24>] kernel_init_freeable+0xdc/0x238
[    1.696394] [<ffff000008b99040>] kernel_init+0x10/0x100
[    1.701670] [<ffff0000080836c0>] ret_from_fork+0x10/0x50
[    1.707037] Unable to handle kernel NULL pointer dereference at virtual address 00000000

soby-mathew, 2018-09-05:
Hi @pangupta 
My first impression looking at the `pwr_domain_on_finish`  is that you are programming the interconnect for every core wakeup which seems wrong. Usually the interconnect will be programmed only for the first core to wakeup in a cluster.

```
static void css_pwr_domain_on_finisher_common(
		const psci_power_state_t *target_state)
{
	assert(CSS_CORE_PWR_STATE(target_state) == ARM_LOCAL_STATE_OFF);

	/* Enable the gic cpu interface */
	plat_arm_gic_cpuif_enable();

	/*
	 * Perform the common cluster specific operations i.e enable coherency
	 * if this cluster was off.
	 */
	if (CSS_CLUSTER_PWR_STATE(target_state) == ARM_LOCAL_STATE_OFF)
		plat_arm_interconnect_enter_coherency();
}
```

As seen from the code above, the interconnect is only programmed if the Cluster state was marked as `OFF`. Cluster state will be OFF only for the first core powering ON in the cluster. The remaining cores of the cluster when powered ON later will find the Cluster state is already `RUNNING` and hence will not do the interconnect programming.

Hope that helps.

pangupta, 2018-10-26:
Still this issue is un-resolved for me.
Included all the RN-F to the Snoop Domain in one go.

closed 2018-10-26 by pangupta