ARM-software/tf-issues #625: What should bl32_init return?

glneo, 2018-09-11:
The BL32 initialization function bl32_init does not seem to have a well defined return value for success. A check was added in [0] that prints a warning if the return is not 0 [1]. Yet the bl32_init for most bl32_init failure case is when the return value is 0 [2][3][4]. So what is the correct failure return code?

[0] https://github.com/ARM-software/arm-trusted-firmware/commit/c9512bca3b69dcb50ac839ce1f6d24a68be46986
[1] https://github.com/ARM-software/arm-trusted-firmware/blob/master/bl31/bl31_main.c#L111
[2] https://github.com/ARM-software/arm-trusted-firmware/blob/master/services/spd/opteed/opteed_main.c#L176
[3] https://github.com/ARM-software/arm-trusted-firmware/blob/master/services/spd/tspd/tspd_main.c#L262
[4] https://github.com/ARM-software/arm-trusted-firmware/blob/master/services/spd/tlkd/tlkd_main.c#L125

antonio-nino-diaz-arm, 2018-09-11:
You are right, the check in [1] in bl31_main() has to be fixed.

antonio-nino-diaz-arm, 2018-09-11:
You can create a PR with the fix. If not, I'll try to do it tomorrow (or someone else will).

glneo, 2018-09-11:
Half of the users return 0, the other !0, so we don't need to fix the check, we need to fix all the handlers that return non-zero as success. I would not know how to do without the help of the SPD authors.

antonio-nino-diaz-arm, 2018-09-18:
I've created two PRs for this: https://github.com/ARM-software/arm-trusted-firmware/pull/1576 and https://github.com/ARM-software/arm-trusted-firmware/pull/1577

closed 2018-09-28 by antonio-nino-diaz-arm