ARM-software/tf-issues #626: BL31 arg0 potentially clobbered by init_c_runtime?

grqpnegre, 2018-09-14:
In arm_bl31_early_plaform_setup for LOAD_IMAGE_V2, a portion of the BL2 address space is copied into BL31's:
       /*
	 * Copy BL33 and BL32 (if present), entry point information.
	 * They are stored in Secure RAM, in BL2's address space.
	 */
As I understand, BL31 NOBITS is overlapping part or all BL2.  Are there any guarantees that this memory won't be clobbered when BL31 performs init_c_runtime near its cold boot entry-point?

soby-mathew, 2018-09-20:
Hi @grqpnegre,

The commit [c099cd3](https://github.com/ARM-software/arm-trusted-firmware/commit/c099cd3942fb163a7894536c1ebaf4ffbf677a82) allowed BL31 to overlay BL2 for ARM platforms to enable more space for BL31. Currently the bl_params list in BL2 data section does not get corrupted by BL31 overlay, but there needs to be assertion check that BL31 will not clobber this region.

Thanks for bringing this to notice.

soby-mathew, 2018-09-20:
Internal ref: https://jira.arm.com/browse/GENFW-2661

closed 2019-01-25 by antonio-nino-diaz-arm