ARM-software/tf-issues #627: Poplar cannot find FIP, stops after BL1 initialization

gsora, 2018-09-17:
I'm having some troubles getting my Poplar board to boot anything.

I've followed the [Poplar Debian build guide](https://github.com/96boards-poplar/Documentation/blob/v1.1/debian/debian_build_instructions.md#flash-images-onto-the-poplar-board-emmc), everything went smoothly, but when I reset the board to make it load from the eMMC, the boot process stops with this error:

```
Bootrom start
Boot Media: eMMC
Decrypt auxiliary code ...OK

lsadc voltage min: 000000FE, max: 000000FF, aver: 000000FE, index: 00000000

Entry boot auxiliary code

Auxiliary code - v1.00
DDR code - V1.1.2 20160205
Build: Mar 24 2016 - 17:09:44

Reg Version:  v134
Reg Time:     2016/03/18 09:44:55
Reg Name:     hi3798cv2dmb_hi3798cv200_ddr3_2gbyte_8bitx4_4layers.reg

Boot auxiliary code success
Bootrom success

LOADER:  Built 2018/09/15 13:07:10+0200 Commit-id df5706a
LOADER:  Switched to aarch64 mode
LOADER:  Entering ARM TRUSTED FIRMWARE
LOADER:  CPU0 executes at 0x0200e000

NOTICE:  Booting Trusted Firmware
NOTICE:  BL1: v1.5(release):v1.6-rc0-4-ge52ed092
NOTICE:  BL1: Built : 13:07:06, Sep 15 2018
ERROR:   error initializing fip
ERROR:   Failed to load BL2 firmware.
```

As per my understanding (which is extremely basic) when ATF is compiled without the `POPLAR_RECOVERY` make flag, it tries to load BL2 from the eMMC.

In the file `plat/hisilicon/poplar/bl1_plat_setup.c`, in the `bl1_plat_setup()` function, when `POPLAR_RECOVERY` is not defined, I should see in the serial terminal an INFO message saying "BL1: initializing emmc", which never appears.

Also, it is impossible to compile with `POPLAR_RECOVERY` defined without commenting line 95 in `plat/hisilicon/poplar/bl1_plat_setup.c` and line 336 in `plat/hisilicon/poplar/bl2_plat_setup.c`.

gsora, 2018-09-17:
Pinging @shawnguo2 for knowledge, since you're maintaining Poplar.

gsora, 2018-09-17:
Looks like commit `0d8052a4eacc73aa808bf4b242f9f64b62875b9d` works, master still fails but with a backtrace now:

```
Bootrom start
Boot Media: eMMC
Decrypt auxiliary code ...OK

lsadc voltage min: 000000FE, max: 000000FF, aver: 000000FE, index: 00000000

Entry boot auxiliary code

Auxiliary code - v1.00
DDR code - V1.1.2 20160205
Build: Mar 24 2016 - 17:09:44

Reg Version:  v134
Reg Time:     2016/03/18 09:44:55
Reg Name:     hi3798cv2dmb_hi3798cv200_ddr3_2gbyte_8bitx4_4layers.reg

Boot auxiliary code success
Bootrom success

LOADER:  Built 2018/09/17 18:38:05+0200 Commit-id df5706a
LOADER:  Switched to aarch64 mode
LOADER:  Entering ARM TRUSTED FIRMWARE
LOADER:  CPU0 executes at 0x0200e000

INFO:    BL1: 0x2016000 - 0x201c000 [size = 24576]
NOTICE:  Booting Trusted Firmware
NOTICE:  BL1: v1.5(debug):v1.6-rc0-4-ge52ed092
NOTICE:  BL1: Built : 18:37:35, Sep 17 2018
INFO:    BL1: RAM 0x2016000 - 0x201c000
INFO:    BL1: cortex_a53: CPU workaround for 855873 was applied
INFO:    BL1: initializing emmc
INFO:    BL1: Loading BL2
INFO:    Using eMMC
ERROR:   error initializing fip
ASSERT: plat/hisilicon/poplar/plat_storage.c:212
BACKTRACE: START: assert
0: EL3: 0x2011024
1: EL3: 0x2012278
2: EL3: 0x20104c0
3: EL3: 0x2010d00
4: EL3: 0x20106b0
5: EL3: 0x200e090
BACKTRACE: END: assert
```

shawnguo2, 2018-09-18:
@gsora I am on travel this week, and will look into it when I get back to my desktop.

gsora, 2018-09-27:
Okay, thank you for your help.

Let me know if you need any information from my board.

shawnguo2, 2018-09-28:
@gsora I linked the issue to the fixes I created as below.
https://github.com/ARM-software/arm-trusted-firmware/pull/1592
https://github.com/ARM-software/arm-trusted-firmware/pull/1593

gsora, 2018-09-28:
I just finished compiling and installing Debian + OP-TEE on my Poplar with your fork, everything works!

Closing because the issue has been fixed.

closed 2018-09-28 by gsora