ARM-software/tf-issues #630: Question: Why should DYN_DISABLE_AUTH only be used for development systems and LOAD_IMAGE_V2?

iq250vip, 2018-09-24:
I understand that in the current implementation, DYN_DISABLE_AUTH (and thus dynamically deciding to call dyn_disable_auth() ) is only used in LOAD_IMAGE_V2 where you have the ability to read the TB_FW_CONFIG file, read the settings there and decide to turn of authentication or not.
I also understand that in this configuration, it should only be used for development systems as it is fairly easy to swap the TB_FW_CONFIG file with a different one which disables authentication.
But this doesn't mean that this is the only use case for DYN_DISABLE_AUTH. In my case, our platform uses a fuse on the chip to dynamically decide that if we should be doing trusted board boot or not, and I'm thinking of compiling with TRUSTED_BOARD_BOOT=1 in my code, and rely on dynamically calling dyn_disable_auth() at runtime when deciding not to do trusted board boot. Is there anything I'm missing here?
Also, in my use case. I don't really need to read an additional "config" file, I'm reading a fuse value in the chip which should always be considered trusted, so I see no point to restrict the DYN_DISABLE_AUTH functionality to LOAD_IMAGE_V2.

iq250vip, 2018-09-24:
Thanks!
But this only solves part of my question. The other part is:
Why should DYN_DISABLE_AUTH only be used for development systems? Am I missing something here?
The current implementation should only be for development systems, but this doesn't prevent it from being used in a production system where I'm reading the embedded fuse in the chip to decide to do trusted board boot or not, right?

soby-mathew, 2018-09-25:
Hi @iq250vip 
> I also understand that in this configuration, it should only be used for development systems as it is fairly easy to swap the TB_FW_CONFIG file with a different one which disables authentication.

On ARM reference platforms, TB_FW_CONFIG will authenticated first and only then will it be used for the decision to disable autheitication or not. Hence , authentication can only be disabled for images following successful authentication of TB_FW_CONFIG. 

> Why should DYN_DISABLE_AUTH only be used for development systems? Am I missing something here?

The `dyn_disable_auth()` function is providing a back door to disable authentication and whether an attack can utilize the backdoor depends the platform. Hence the reason to wrap this under the build flag and disabling it by default.

The platform can choose to enable the build flag even on production system if it deems that mechanism to disable authentication is secure enough.

Hope that clarifies.

iq250vip, 2018-09-25:
Thanks! This cleared up my question.

For the ARM reference platforms, even though TB_FW_CONFIG have to be authenticated first. If there ever existed a TB_FW_CONFIG which disabled authentication, I see no way of preventing copying of this file (along with it's certificates) to another platform using the same keys to also disable that from doing authentication for later images.

soby-mathew, 2019-01-16:
Closing the issue.

closed 2019-01-16 by soby-mathew