ARM-software/tf-issues #631: Broken rules in tools/doimage and tools/stm32image/Makefile

jan-kiszka, 2018-10-01:
Too lazy to fork for this trivial patch (with critical impact, depending on the build environment):

```
From: Jan Kiszka <jan.kiszka@siemens.com>
Date: Mon, 1 Oct 2018 23:55:09 +0200
Subject: [PATCH] tools: Fix broken object compilation rules

As these rules depend on non-existing headers as well (likely copy &
pasted from fiptool), they never matched, and the built-in rules were
used. That led to random breakages when e.g. CPPFLAGS was suddenly
evaluated and contained invalid options.

---
 tools/doimage/Makefile    | 2 +-
 tools/stm32image/Makefile | 2 +-
 2 files changed, 2 insertions(+), 2 deletions(-)

diff --git a/tools/doimage/Makefile b/tools/doimage/Makefile
index bc74369f..9cec681a 100644
--- a/tools/doimage/Makefile
+++ b/tools/doimage/Makefile
@@ -39,7 +39,7 @@ ${PROJECT}: ${OBJECTS} Makefile
 	@echo "Built $@ successfully"
 	@echo
 
-%.o: %.c %.h Makefile
+%.o: %.c Makefile
 	@echo "  CC      $<"
 	${Q}${CC} -c ${CFLAGS} ${INCLUDE_PATHS} $< -o $@
 
diff --git a/tools/stm32image/Makefile b/tools/stm32image/Makefile
index 80dfbecf..07e26054 100644
--- a/tools/stm32image/Makefile
+++ b/tools/stm32image/Makefile
@@ -39,7 +39,7 @@ ${PROJECT}: ${OBJECTS} Makefile
 	@echo "Built $@ successfully"
 	@${ECHO_BLANK_LINE}
 
-%.o: %.c %.h Makefile
+%.o: %.c Makefile
 	@echo "  CC      $<"
 	${Q}${CC} -c ${CFLAGS} $< -o $@
 
-- 
2.16.4
```

antonio-nino-diaz-arm, 2018-10-01:
@Yann-lms @kostapr

Yann-lms, 2018-10-02:
If I apply this patch, I get errors about getopt and optarg undefined. I will check what's wrong.

jan-kiszka, 2018-10-02:
v2:
```
From: Jan Kiszka <jan.kiszka@siemens.com>
Date: Tue, 2 Oct 2018 09:28:00 +0200
Subject: [PATCH] tools: Fix broken object compilation rules

As these rules depend on non-existing headers as well (likely copy &
pasted from fiptool), they never matched, and the built-in rules were
used. That led to random breakages when e.g. CPPFLAGS was suddenly
evaluated and contained invalid options.

For the stm32image, this reveals that we were relying on the built-in
rules by passing -D_GNU_SOURCE via CPPFLAGS, rather than using CFLAGS as
used in the local rule. Fix that as well.

---
 tools/doimage/Makefile    | 2 +-
 tools/stm32image/Makefile | 5 ++---
 2 files changed, 3 insertions(+), 4 deletions(-)

diff --git a/tools/doimage/Makefile b/tools/doimage/Makefile
index bc74369f..9cec681a 100644
--- a/tools/doimage/Makefile
+++ b/tools/doimage/Makefile
@@ -39,7 +39,7 @@ ${PROJECT}: ${OBJECTS} Makefile
 	@echo "Built $@ successfully"
 	@echo
 
-%.o: %.c %.h Makefile
+%.o: %.c Makefile
 	@echo "  CC      $<"
 	${Q}${CC} -c ${CFLAGS} ${INCLUDE_PATHS} $< -o $@
 
diff --git a/tools/stm32image/Makefile b/tools/stm32image/Makefile
index 80dfbecf..a593d316 100644
--- a/tools/stm32image/Makefile
+++ b/tools/stm32image/Makefile
@@ -12,8 +12,7 @@ PROJECT := stm32image${BIN_EXT}
 OBJECTS := stm32image.o
 V := 0
 
-override CPPFLAGS += -D_GNU_SOURCE
-CFLAGS := -Wall -Werror -pedantic -std=c99
+CFLAGS := -Wall -Werror -pedantic -std=c99 -D_GNU_SOURCE
 ifeq (${DEBUG},1)
   CFLAGS += -g -O0 -DDEBUG
 else
@@ -39,7 +38,7 @@ ${PROJECT}: ${OBJECTS} Makefile
 	@echo "Built $@ successfully"
 	@${ECHO_BLANK_LINE}
 
-%.o: %.c %.h Makefile
+%.o: %.c Makefile
 	@echo "  CC      $<"
 	${Q}${CC} -c ${CFLAGS} $< -o $@
 
-- 
2.16.4
```

antonio-nino-diaz-arm, 2018-10-05:
Fixed in https://github.com/ARM-software/arm-trusted-firmware/pull/1609

closed 2018-10-05 by antonio-nino-diaz-arm