ARM-software/tf-issues #632: Trusted Firmware-A v2.0 now available

joannafarley-arm, 2018-10-02:
Trusted Firmware version 2.0 is now available and can be found here:
https://github.com/ARM-software/arm-trusted-firmware/tree/v2.0

This release removes a number of deprecated API's otherwise it is unchanged from 1.6 release.

Please refer to the [readme](https://github.com/ARM-software/arm-trusted-firmware/blob/master/readme.rst) and [change log](https://github.com/ARM-software/arm-trusted-firmware/blob/master/docs/change-log.rst) for further information.

Regards

Joanna

kettenis, 2018-10-21:
FWIW, Rockchip RK3399 seems to be in working order in 2.0. OpenBSD runs fine on my Firefly-RK3399 board with Trusted Firmware version 2.0.

antonio-nino-diaz-arm, 2018-10-22:
@kettenis thanks for telling us!