ARM-software/tf-issues #635: Questions about GPT

Yann-lms, 2018-10-03:
Hi,

I'd like to use GPT for the MMC devices on STM32MP1 platform.
For the moment, I have a proprietary solution, not upstream-able as-is.
But I've seen that there is a GPT support in TF-A (in drivers/partition/), so I'd like to use it if possible.
The thing is there seem to be no platform in the tree that use this GPT implementation.

Are there any out-of-tree platform using it?

Or, to ease my port, are there any code examples that use this?

Maybe @hzhuang1, you have such code, as you've made the GPT code?

Thanks,
Yann

hzhuang1, 2018-10-04:
The partition driver was planned to use on HiKey platform. Since it relies on USB driver for the flushing feature, it'll only make ATF too complex. I move the USB and flushing feature into another project, atf-fastboot, so partition driver isn't really used in the ATF. But the code isn't same exactly.

I remember that the partition driver was tested before merged. You could make use of the APIs to get the partition entry by name.

1. Call partition_init(image_id) in initial.
2. Call get_partition_entry(name) to get the partition entry. In this entry, we could know the start and length properties of the partition.

Yann-lms, 2018-10-04:
Thanks for your answer.
I've seen one of your old branches, where you were using that for BL2U.
I've been able to use the partition_init and dump the partition list.
I'll continue the implementation.
Question is answered, I can close the ticket.

closed 2018-10-04 by Yann-lms