ARM-software/tf-issues #636: Why not support backtrace in TF

puckchen, 2018-10-08:
Hello,
    TF debug is quite diffcult, panic without useful information.
why not support backtrace for it.

antonio-nino-diaz-arm, 2018-10-08:
There is already a backtrace function that is called from ``assert()`` and ``panic()``. However, it doesn't work during an actual crash (segmentation fault or similar). Is this what you need?

puckchen, 2018-10-08:
It doesn't have the function call graph like linux kernel which is really useful for debug.

antonio-nino-diaz-arm, 2018-10-08:
Well, that would use a lot of memory that many platforms don't have. Actually, we even removed the function names from the asserts a long time ago for this reason: https://github.com/ARM-software/arm-trusted-firmware/pull/936

antonio-nino-diaz-arm, 2018-10-08:
There is another problem, though. We remove all debug symbols, so we need to add the function names and addresses to the binary somehow.

puckchen, 2018-10-08:
> There is another problem, though. We remove all debug symbols when we build, so we need to add the function names and addresses to the binary somehow.

Only need the elf itself.
Take a look the kernel solution, scripts/kallsyms.c to generate symbol arrary.
And kernel/kallsyms.c to get function name from the generated arrary.

antonio-nino-diaz-arm, 2018-10-09:
The problem is that then you have to somehow add the symbols to the binary, which is not straightforward. In general, it's quite a bit of effort for not too much gain. The idea was to parse the output with a script that takes the elf file as input and translates the addresses to lines of code.

puckchen, 2018-10-09:
> The problem is that then you have to somehow add the symbols to the binary, which is not straightforward. In general, it's quite a bit of effort for not too much gain. The idea was to parse the output with a script that takes the elf file as input and translates the addresses to lines of code.

Make it more friendly to ourselfs, I can see it's a great work.

puckchen, 2018-10-16:
@soby-mathew  I already done and tested it on hikey960. I reuse the kernel kallsyms.c. But the code is a GPL license, so it's a pity story. I know that you will no merge it.

antonio-nino-diaz-arm, 2018-10-16:
@puckchen yes, unfortunately we can't accept GPL code.