ARM-software/tf-issues #637: How to use U-Boot on a rpi3

hschmidbauer, 2018-10-12:
I'd like to build a complete trusted boot (for educationnal purpose) on a raspberry pi 3 B. However I can't get the ATF to load U-Boot as bootloader (not getting past BL31 when it tries to get to BL33).
Can someone explain me how to do it ?

For now I'm building my ATF with this command : 
```
    CROSS_COMPILE=aarch64-linux-gnu- make PLAT=rpi3              \
    PRELOADED_BL33_BASE=0x02000000                                          \
    RPI3_PRELOADED_DTB_BASE=0x01000000                                   \
    RPI3_DIRECT_LINUX_BOOT=1                                                        \
    RPI3_BL33_IN_AARCH32=1					                           \
    GENERATE_COT=1                                                                          \
    TRUSTED_BOARD_BOOT=1
```

And replacing the kernel8.img file with the u-boot.bin file 

U-Boot is built using this command : 
```
make rpi_3_defconfig
make CROSS_COMPILE=aarch64-linux-gnu-
```

antonio-nino-diaz-arm, 2018-10-12:
This is one of the things I wanted to get properly working, yes... I know that it can be done:

https://github.com/OP-TEE/manifest/blob/da2ce6baa0cc599998f62d2cd163520125517771/rpi3.xml
https://github.com/OP-TEE/build/blob/2e947f2ee5ae8d2fa3624f90c893f901187dd0b0/rpi3.mk

However, when I tried to build mainline U-Boot it didn't go well.

As far as I can see, in the second link they are including U-Boot in the FIP as BL33. Maybe you can try the same way they are doing it:

```
CROSS_COMPILE=aarch64-linux-gnu- make PLAT=rpi3 \
GENERATE_COT=1                                  \
TRUSTED_BOARD_BOOT=1                            \
BL33=$(U-BOOT_BIN)
```

In your case you have another problem, though, you are jumping to BL33 in 32-bit mode with ``RPI3_BL33_IN_AARCH32=1``.

hsmb77, 2018-10-12:
Just tried it but nothing changed. Here is my output while booting:
```
INFO:    Entry point address = 0x11000000
INFO:    SPSR = 0x3c9
NOTICE:  Booting Trusted Firmware
NOTICE:  BL1: v1.6(release):v1.6-8-g07a2fd6a
NOTICE:  BL1: Built : 10:11:06, Oct 12 2018
INFO:    BL1: RAM 0x100ee000 - 0x100f9000
INFO:    Using crypto library 'mbed TLS'
NOTICE:  rpi3: Detected: Raspberry Pi 3 Model B (1GB, Sony, UK) [0x00a02082]
INFO:    BL1: Loading BL2
INFO:    Loading image id=6 at address 0x100b4000
INFO:    Image id=6 loaded: 0x100b4000 - 0x100b446f
INFO:    Loading image id=1 at address 0x100b4000
INFO:    Image id=1 loaded: 0x100b4000 - 0x100c43c8
NOTICE:  BL1: Booting BL2
INFO:    Entry point address = 0x100b4000
INFO:    SPSR = 0x3c5
NOTICE:  BL2: v1.6(release):v1.6-8-g07a2fd6a
NOTICE:  BL2: Built : 10:11:09, Oct 12 2018
INFO:    Using crypto library 'mbed TLS'
INFO:    BL2: Doing platform setup
INFO:    BL2: Loading image id 3
INFO:    Loading image id=7 at address 0x100e0000
INFO:    Image id=7 loaded: 0x100e0000 - 0x100e060e
INFO:    Loading image id=9 at address 0x100e0000
INFO:    Image id=9 loaded: 0x100e0000 - 0x100e04da
INFO:    Loading image id=13 at address 0x100e0000
INFO:    Image id=13 loaded: 0x100e0000 - 0x100e0430
INFO:    Loading image id=3 at address 0x100e0000
INFO:    Image id=3 loaded: 0x100e0000 - 0x100e7079
INFO:    BL2: Loading image id 5
INFO:    Loading image id=11 at address 0x11000000
INFO:    Image id=11 loaded: 0x11000000 - 0x110004ea
INFO:    Loading image id=15 at address 0x11000000
INFO:    Image id=15 loaded: 0x11000000 - 0x11000440
INFO:    Loading image id=5 at address 0x11000000
INFO:    Image id=5 loaded: 0x11000000 - 0x1107db78
NOTICE:  BL1: Booting BL31
INFO:    Entry point address = 0x100e0000
INFO:    SPSR = 0x3cd
NOTICE:  BL31: v1.6(release):v1.6-8-g07a2fd6a
NOTICE:  BL31: Built : 10:11:13, Oct 12 2018
INFO:    BL31: Initializing runtime services
INFO:    BL31: Preparing for EL3 exit to normal world
INFO:    Entry point address = 0x11000000
INFO:    SPSR = 0x3c9

```

antonio-nino-diaz-arm, 2018-10-12:
I guess that revision a032e0a6aed208977f48e78d2cc497b91543beaf of U-Boot still fails, right? If so, I will take a look at this. I'm not too surprised it doesn't work, however. I've only tested the direct-to-kernel boot.

hsmb77, 2018-10-15:
Tried again with that revision to be sure but it still doesn't work. Were you already able to have a look at it?

antonio-nino-diaz-arm, 2018-10-15:
No, sorry. The problem is that I'm not exactly an expert in U-Boot. From the point of view of the TF everything is correct, and I don't know what assumptions U-Boot is making. I will talk to the maintainer of the RPi3 port.

Also, even if you managed to get U-Boot to boot, you would lose the TF. U-Boot doesn't know about the memory reserved for the TF, so it may use it, and it won't tell Linux to protect it unless you do it in the command line or modify the DTS file to add the protected region. Normally this isn't needed because the TF runs in memory that U-Boot simply can't see, but the RPi3 is special.

ldts, 2018-10-15:
this is how it was done in hikey in case it helps
https://github.com/ldts/u-boot/blob/rcar_gen3/devel/board/hisilicon/hikey/hikey.c#L447

antonio-nino-diaz-arm, 2018-10-15:
Well, ideally the TF would pass a DT to U-Boot with the reserved regions instead of hardcoding them, and Linux would use the same DT.

ldts, 2018-10-15:
sure that would be the proper way of doing it (most firmwares do that, like Qualcomm's Little Kernel [1] 
[1] https://github.com/ldts/u-boot/commit/9337dfb4342a7bf2164405cdcb9c92b857ea0974#diff-43b64201de88974eddc2a31dbac84eaf

Rusack, 2018-10-17:
Is there a solution ? I'm running into the same problem, tried to compile the ATF  with same options as recommended above but even when including U-Boot as BL33, it does hang when it's supposed to boot U-Boot even if the image is loaded correctly. (Maybe a problem with the U-boot image ? I'm compiling it using the default config for RPI).

My output is pretty much the same, the debug infos aren't helping.
```
NOTICE:  Booting Trusted Firmware
NOTICE:  BL1: v1.6(debug):v1.6-8-g07a2fd6a
 for 843419 was applied
INFO:    BL1: c018
INFO:    BL1: RAM 0x100ee000 - 0x100f7000
INFO:    BL1: cortex_a53: CPU workaround for 843419 was applied
INFO:    BL1: cortex_a53: CPU workaround for 855873 was applied
NOTICE:  rpi3: Detected: Raspberry Pi 3 Model B (1GB, Sony, UK) [0x00a02082]
INFO:    BL1: Loading BL2
INFO:    Loading image id=1 at address 0x100b4000
INFO:    Image id=1 loaded: 0x100b4000 - 0x100b91f1
NOTICE:  BL1: Booting BL2
INFO:    Entry point address = 0x100b4000
INFO:    SPSR = 0x3c5
NOTICE:  BL2: v1.6(debug):v1.6-8-g07a2fd6a
NOTICE:  BL2: Built : 10:44:08, Oct 17 2018
INFO:    BL2: Doing platform setup
INFO:    BL2: Loading image id 3
INFO:    Loading image id=3 at address 0x100e0000
INFO:    Image id=3 loaded: 0x100e0000 - 0x100e8081
INFO:    BL2: Loading image id 5
INFO:    Loading image id=5 at address 0x11000000
INFO:    Image id=5 loaded: 0x11000000 - 0x110707b0
NOTICE:  BL1: Booting BL31
INFO:    Entry point address = 0x100e0000
INFO:    SPSR = 0x3cd
NOTICE:  BL31: v1.6(debug):v1.6-8-g07a2fd6a
NOTICE:  BL31: Built : 10:44:13, Oct 17 2018
INFO:    BL31: Initializing runtime services
INFO:    BL31: cortex_a53: CPU workaround for 843419 was applied
INFO:    BL31: cortex_a53: CPU workaround for 855873 was applied
INFO:    BL31: Preparing for EL3 exit to normal world
INFO:    Entry point address = 0x11000000
INFO:    SPSR = 0x3c9
```

antonio-nino-diaz-arm, 2018-10-17:
I think it's worth asking the maintainer of the port of U-Boot to the Raspberry Pi Alexander Graf. https://github.com/u-boot/u-boot/blob/master/board/raspberrypi/rpi/MAINTAINERS

antonio-nino-diaz-arm, 2018-10-17:
Ok, so this is the first step: https://github.com/ARM-software/arm-trusted-firmware/pull/1637

antonio-nino-diaz-arm, 2018-10-19:
By the way, if you want to at least get U-Boot to get to the prompt, try changing the following function and configuration option:

**board/raspberrypi/rpi/rpi.c**
```
unsigned long board_get_usable_ram_top(unsigned long total_size)
{
        return 0x3F000000;
}
```

**configs/rpi_3_defconfig**
```
CONFIG_SYS_TEXT_BASE=0x11000000
```

I haven't actually tried to boot Linux from the prompt, though.

mjeveritt, 2020-06-01:
Hiya, what's the current status of this, as I'm planning to deploy this configuration on a new Pi3A I've bought recently.

I use Gentoo Linux here, so used to compiling everything from scratch - is there any useful testing I can do as part of the process?

Thanks in anticipation! :smile: