ARM-software/tf-issues #641: GIC redistributor base address 0x10000 is wrong . update it with correct value

mariusxp, 2018-10-24:
The 0x10000 offset from GIC_BASE are Distributor registers for message based SPIs. These registers are aliases of register also addressed in the first page. Write to this page to generate SPIs. Reg prefix GICD_

The start offset for redistributor should be at 0x90000 .
GIC Redistributor SGI and PPI registers for core 0.  SGI_base 0 frame. Reg prefix GICR_
...
You have to update gicv3.h file.

jts-arm, 2018-10-30:
Hello @mariusxp,

I'm not sure which macro definitions you refer to, exactly, neither what is the problem you are trying to raise. Could you please specify a bit more? The base addresses of the various GIC components (distributor, redistributors etc.) are platform dependent and, as such, cannot be defined in the gicv3.h file.

Of course, if you believe something has to be changed you are welcome to submit a patch with the proposed changes.

Thanks,
John

soby-mathew, 2019-01-16:
The query is answered. Please reopen if you think the issue is unresolved.

closed 2019-01-16 by soby-mathew