ARM-software/tf-issues #642: CCN-508: Is there any API in CCN driver to set the Node's Auxilary Control Register

pangupta, 2018-10-26:
CCN-508: Is there any API in CCN driver to set the Node's Auxiliary Control Register based on the Node type and Node Identifier?

For same SoC, there can be different use cases, where it can be deployed. 
Depending on the deployment scenario, there is requirement to fine tune some of the ARM provided registers like Auxiliary control register.

Is there any API in CCN driver to set the Node's Auxiliary Control Register based on the Node type and Node Identifier?

soby-mathew, 2018-10-30:
Currently there is no API to do this AFAICS. Please submit a PR to do the same and we can discuss further based on that.

pangupta, 2018-11-02:
Raised the PR (https://github.com/ARM-software/arm-trusted-firmware/pull/1664).

antonio-nino-diaz-arm, 2018-12-05:
Fixed in https://github.com/ARM-software/arm-trusted-firmware/pull/1679

closed 2018-12-05 by antonio-nino-diaz-arm