ARM-software/tf-issues #645: GIC v3 lowest priority valuse  GIC_LOWEST_NS_PRIORITY		U(0xfe) can be wrong

mariusxp, 2018-11-05:
For me it works as expected if I chose for this values 0xF0.
(the lowest applicable interrupt priority is 0xf0, since the Gic takes (took) 5 msb bits only into account).
0xf1 and above - this is the IDLE priority, and the ISRs are not triggered if this priority is used.
 Since the priority mask in the A53 is at the minimum, also 0x1F, these interrupts aren't signalled to the core as the interrupt priority must exceed the priority mask value.  All

jeenu-arm, 2018-11-07:
Yes, you're correct that the lowest priority values for both Secure and Non-secure interrupts is implementation-defined. The constants in `gic_common.h` are meant to enumerate GIC architectural values. For your purpose, you should use either a platform-defined macro, or discover the value at run time.

To avoid confusion on this, since the lowest priority macros are not used in generic code nor any upstream platforms, I think we'll remove it altogether.

soby-mathew, 2018-12-18:
Fixed by https://github.com/ARM-software/arm-trusted-firmware/pull/1720

closed 2018-12-18 by soby-mathew