ARM-software/tf-issues #646: MULTI_CONSOLE_API behavior in crash context is broken

jwerner-chromium, 2018-11-06:
@soby-mathew @satheesbalya-arm 

Commit https://github.com/ARM-software/arm-trusted-firmware/commit/d35cc347d changed the main MUTLI_CONSOLE_API framework functions to use callee-saved registers and save the caller's registers on the stack, rather than reserving specific registers (which are caller-saved according to AAPCS) from use by the console drivers it calls. Unfortunately, this completely breaks its behavior in crash context, since TF does not make a stack available in that environment (in fact do_crash_reporting() uses SP as a scratch register).

One of the main design goals of the MULTI_CONSOLE_API was to unify the normal and crash consoles, so that consoles only need to be set up once by the platform and will automatically also work to output register dumps in case of a crash. This allows platforms much more flexible crash reporting than overriding plat_crash_console_xxx functions directly would have (e.g. it allows using the cbmem_console memory console driver for crash dumps, which has to be initialized dynamically with information passed in by an earlier boot stage and thus cannot be reinitialized from hardcoded values in a crash situation).

I've confirmed this issue on an RK3399 device. Crash output is completely broken on the current master branch, but after reverting this commit it works fine again. I personally think returning to the previous version of the code would be the best option here... it tries to strike a balance between not making the ABIs between different assembly functions too confusing, while still making sure the external API to its callers is AAPCS compliant so that it can be called from C code and at the same time not relying on a stack. If we really feel like this is too brittle, I think the only real way to fix it would be to set up a real exception stack for use by all crash handlers... however, this may come with its own drawbacks in case a platform crash handler wants to avoid memory accesses completely, and it may also be a larger effort. In the meantime, if there are no objections, I would like to revert that commit to restore the previous functionality since this is an important feature for Chrome OS.

antonio-nino-diaz-arm, 2018-11-07:
As I've mentioned in the thread in https://github.com/ARM-software/arm-trusted-firmware/pull/1635#discussion_r231429834, the main problem is that reverting that commit would make it impossible to have C console drivers. warp7 has a C driver for regular text output and a simpler assembly driver for crash output, which is fair.

https://github.com/ARM-software/arm-trusted-firmware/blob/9c10e3485b10ece95f9af8a6af79558397ee1d78/drivers/imx/uart/imx_uart.c

https://github.com/ARM-software/arm-trusted-firmware/blob/9c10e3485b10ece95f9af8a6af79558397ee1d78/drivers/imx/uart/imx_crash_uart.S

soby-mathew, 2018-11-07:
Suggested a solution in https://github.com/ARM-software/arm-trusted-firmware/pull/1635#discussion_r231429834

Copying the suggestion here: 

> It is a an useful capability to allow platforms to implement their console drivers in C, as long as they don't use it for crash reporting which is what the patch is allowing to do.

> @antonio-nino-diaz-arm and I have had a discussion and we think the solution here is to have a new API crash_console_putc which is equivalent console_putc but doesn't use the stack. The default plat_crash_console_putc will call into crash_console_putc. The implementation of crash_console_putc can be simpler than console_putc because it doesn't need to check for the
console_state as it is always invoked in the crash state. This means the plat_crash_console_init can be empty. Similarly crash_console_flush can be implemented.

> Please let us know of your opinion.

jwerner-chromium, 2018-11-07:
> The implementation of crash_console_putc can be simpler than console_putc because it doesn't need to check for the console_state as it is always invoked in the crash state

You could skip the console_switch_state() call with this, but I think that's not really a problem anyway (it's already AACPS compliant as is). You would still need to check each individual console for whether it has set the CONSOLE_FLAG_CRASH flag, if we want to retain the capability that platforms can decide which consoles should be used for crash reporting and which shouldn't. And anyway, I don't think removing that check really saves you a register, so it doesn't solve the fundamental problem.

But we could have two implementations of console_putc and console_flush (one that uses the stack and one that expects the functions it calls to preserve certain registers), yes. It would mean we'd essentially have to duplicate that code and maintain it twice, and it's already somewhat complicated assembly logic so I think it would be nice to avoid that.

I honestly don't really understand the value of using C functions for console drivers, particularly on the example of the IMX driver, since they have implemented an assembly version for crash reporting anyway and it seems to be doing almost all the same stuff as the C version. Don't you just have to maintain twice as much code now? I understand that writing and maintaining C is easier than writing and maintaining assembly, but if you're gonna write and maintain the assembly anyway, I'd expect it is easier to use that assembly version for both the normal and crash console rather than having a second copy in C. I think having console drivers in C would only make sense if you're willing to completely give up on crash reporting with that console.

There's also the fact that putc/getc/flush implementations are usually very simple, and if there's anything about a console driver that's really too complicated for assembly it's usually the init function. In the MULTI_CONSOLE_API, the driver's register() function calls into the framework but not the other way around, so it would be okay for a console driver to implement the initialization code in C and only the putc/getc/flush callbacks in assembly as long as they understand that they can't register a new console from a crash context in that case. Would that maybe be a reasonable compromise?

Alternatively, if you really think it needs to be possible to implement all console callbacks in C, I would suggest maybe using a little wrapper stub to save the necessary registers from within the console driver callback, before it calls the C function. I think this might be preferable to adding more complexity to the main framework code. You could define a helper macro like this so that driver implementations don't really need to write any assembly themselves, just call the macro at the top of their driver file:

    #define DECLARE_C_CONSOLE_DRIVER(name) \
        asm ("console_"##name##"_putc_wrapper:\n\t" \
            "stp x30, x15, [sp, #-16]!\n\t" \
            "stp x14, x13, [sp, #-16]!\n\t" \
            "bl console_"##name##"_putc\n\t" \
            "ldp x14, x13, [sp], #16\n\t" \
            "ldp x30, x15, [sp], #16\n\t" \
            "ret\n\t"); \
        ...same for getc and flush...

Alternatively, you could also use make's target-specific variable mechanism to add -fcall-saved-x13 -fcall-saved-x14 -fcall-saved-x15 to CFLAGS for that console driver file only. The console driver could provide its own .mk file that platforms can just include, so you don't have to know about this in every individual platform.mk file using it. We could even create a make function to wrap this so driver implementers just need to put something like `$(call c_console_driver_file,mydriver.c)` into that mydriver.mk file and don't need to understand how target-specific variables work in detail.

> [From the other thread]
> What I meant is rework the to crash reporting to use the higher register set, and the MULTI_CONSOLE stays as it is. The MULTI_CONSOLE API should remain AAPCS compliant.

I think this would be a good idea independently of all this, because it would just allow console driver implementations that are in assembly to use more registers. Right now putc is constrained to only two scratch registers (four if you use x16 and x17), which can be pretty tight for more complicated drivers. But this doesn't really help with implementing console drivers in C.

antonio-nino-diaz-arm, 2018-11-08:
The problem is that as long as you have one single console driver in C you need to make  ``console_putc`` and ``console_flush`` AACPS compliant. And it is not a bad idea to have a driver written in C, it's more maintainable and portable: you don't have to write the same code for AArch32 and AArch64, for example. If someone decides to have a C driver for one console and an assembly driver for another console that is used for crash output, that looks perfectly fine by me, regardless of the IMX example.

Also, the fact that making the functions AACPS compliant broke the crash reporting code makes me think that there is something wrong with the crash reporting code, and that we should be fixing that rather than hacking the multi console driver and limiting what can be done with it.

The ``console_putc`` and ``console_flush`` functions aren't too complicated, and not having to check the global console state (as we know we are in crash mode) would make ``console_crash_putc`` and ``console_crash_flush`` slightly simpler. I don't think they will be a particularly big maintenance problem. I personally think that your C macro with inline assembly code or passing C flags to build that specific file is more error-prone than duplicating the functions, and a bigger maintenance burden. Modifying the build system looks also like a bad idea, it's hacky enough as it is right now, and we are looking into replacing it by something better (which will take time).

In my opinion the crash reporting code should point the stack pointer somewhere so that it can be used as emergency stack and then assume that the stack can be used. If that doesn't work something has gone seriously wrong and the information you may get in the crash output is probably meaningless anyway. This would be tricky to fix anyway.

I think the cleanest solution is to duplicate the functions.

jwerner-chromium, 2018-11-09:
Fair enough. I agree that all of these options will solve the problem in one way or another, so we can go with whatever you think works best.

How are we going to proceed from here then? Is someone from Arm going to work on this fix? I'd prefer if this doesn't stay broken for too long, because it is an important feature for us. If we're not sure when someone will be able to get around to this, maybe it would be better to revert the breaking commit for now, and then it can be reapplied together with that redesign once it's ready. As far as I can tell no C console driver has actually switched to using MULTI_CONSOLE_API yet, so reverting this wouldn't break anything at the moment... it would just make clear that this issue needs to be solved before any C console drivers can be ported.

soby-mathew, 2018-11-09:
Hmm, it seems using MULTI-CONSOLE API with crash console was never correct.

https://github.com/ARM-software/arm-trusted-firmware/blob/master/bl31/aarch64/crash_reporting.S#L216

We assume that all the registers except x0 - x6, x30 will be preserved. Invoking console_putc() from MULTI_CONSOLE_API will corrupt the higher registers (x12 - x15) which is still not saved at that point.

Overriding the crash_console_* seems to be only correct solution for now. Need more thoughts on this.

jwerner-chromium, 2018-11-14:
> We assume that all the registers except x0 - x6, x30 will be preserved. Invoking console_putc() from MULTI_CONSOLE_API will corrupt the higher registers (x12 - x15) which is still not saved at that point.

Crap, yeah... looks like I forgot about that. The mechanism still works but those registers aren't reported correctly. Thanks for pointing it out.

> [...]
> I think this should be a reasonable compromise for using crash reporting with MULTI_CONSOLE_API. Let us know what you think of this.

I think your approach may work with some tweaking (plat_crash_console_init cannot use x5 or x6 due to asm_assert()), although it seems a tad bit complicated to me. If saving space is so important here, do we actually need to have a separate register stash per CPU? Console drivers are usually not that thread-safe anyway... at least the output will definitely be garbled if you try to print from two CPUs at once. If we had two simultaneous crashes and we only had one copy of the crash register stash, the worst that would happen is that some of the printed register values will be wrong in the output that's already going to be garbled anyway. So if space is a prime concern, why not just switch to a single global register stash?

antonio-nino-diaz-arm, 2018-11-15:
> If we had two simultaneous crashes and we only had one copy of the crash register stash, the worst that would happen is that some of the printed register values will be wrong in the output that's [...]

That's a good point. Probably one global stash is enough. It is really unlikely for two CPU to crash at the same time, anyway.

soby-mathew, 2018-11-15:
The CPUs could crash with different cache states, and it is usually bad if 2 CPUs write to same memory location with differing cache states. That's why the crash buffers are allocated on per-cpu data which are cache line aligned.

I didn't realize asm_assert() corrupted x5 and x6. Need to take a detailed look.

jwerner-chromium, 2018-11-15:
> The CPUs could crash with different cache states, and it is usually bad if 2 CPUs write to same memory location with differing cache states.

Yeah, but "bad" as in "the chip may catch on fire" or just "bad" as in "the contents of that memory location and related cache entries may be undefined"? I think in the rare case that you get this, the crash output will already be useless anyway, so the value of those memory locations doesn't matter. They're not used to store return addresses or anything, after all, just register values for dumping. There's no recovery after crashes in Trusted Firmware anyway, nothing will ever read those locations again after the crash dumper.

jwerner-chromium, 2018-11-17:
> There is no recovery, but that doesn't preclude a platform from implementing a UART/debug reporting interface that can output report for each CPU sequentially within the plat_crash_console_*.

Fair enough. Although I think if you wanted that, it would work way better to just implement that mechanism in the common crash reporting framework rather than have platforms implement it themselves. That way, you could still save (at least some of) the crash register stash space by moving it out of cpu_data_t.

> Perhaps plat_crash_console_init() can initialise a global buffer and use that to print. Also set a flag to ensure no other CPUs will print any crash messages.

Yes, I agree that would be the best alternative if you don't want this handled by the common code. Rather than having that aux pointer in cpu_data_t, plat_crash_console_init() can just grab a lock and then plat_crash_console_putc/flush() can use a single global buffer to save/restore those extra registers.

soby-mathew, 2018-11-21:
> I'm not sure if this is just because dcache was disabled

Yes, that's likely to be the case. As I mentioned earlier, spin_locks are not expected to work with cache disabled. Basically crash reporting has to be skipped in these code paths to use the spin_lock.

This is something I have currently which basically bails out if cache is disabled.

```diff
diff --git a/plat/common/aarch64/crash_console_helpers.S b/plat/common/aarch64/crash_console_helpers.S
index 5af8db2..46f72b5 100644
--- a/plat/common/aarch64/crash_console_helpers.S
+++ b/plat/common/aarch64/crash_console_helpers.S
@@ -11,6 +11,8 @@
 
 #include <asm_macros.S>
 #include <console.h>
+#include <platform_def.h>
 
        .globl  plat_crash_console_init
        .globl  plat_crash_console_putc
@@ -18,6 +20,27 @@
 
 #if MULTI_CONSOLE_API
 
+#define NUM_OF_REG_TO_SAVE     8
+#define REG_SZ                 8
+#define AUX_BUF_SIZ    (NUM_OF_REG_TO_SAVE * REG_SZ)
+
+       /*
+        * Define the auxiliary global crash buffer for the
+        * plat_crash_console_putc/flush to stash registers.
+        * Also define a flag `console_used` to prevent other CPUs from
+        * using the crash console once it has been used. The access
+        * to the `console_used` variable is protected by
+        * the `crash_console_lock` spin lock.
+        */
+
+.section .data.aux_crash_buf ; .align 4
+       aux_crash_buf:
+       .space AUX_BUF_SIZ
+.section .data.crash_console_lock ; .align SPINLOCK_ASM_ALIGN
+       define_asm_spinlock crash_console_lock
+.section .data.console_used; .align 3
+       console_used: .word 0x0
+
        /* -----------------------------------------------------
         * int plat_crash_console_init(void)
         * Use normal console by default. Switch it to crash
@@ -30,6 +53,7 @@
         * with an implementation that initializes a console
         * driver with hardcoded parameters. See
         * docs/porting-guide.rst for more information.
+        * This function must only clobber x0 - x3.
         * -----------------------------------------------------
         */
 func plat_crash_console_init
@@ -44,10 +68,36 @@ func plat_crash_console_init
        ret
 #else  /* IMAGE_BL1 */
        mov     x3, x30
-       mov     x0, #CONSOLE_FLAG_CRASH
-       bl      console_switch_state
+
+       /*
+        * Spin locks can be used when SCTLR.C bit is enabled.
+        * Check if SCTLR.C bit is set else return early.
+        */
+       mrs     x1, sctlr_el3
+       tst     x1, #SCTLR_C_BIT
+       beq     skip_crash_console
+
+       /* Acquire lock and check if crash console is used already */
+       adr     x0, crash_console_lock
+       bl      spin_lock
+
+       adr     x1, console_used
+       ldr     x2, [x1]
+       /* Return early if console already used */
+       cbnz    x2, unlock_skip
+       mov     x2, #1
+       str     x2, [x1]
+       bl      spin_unlock
        mov     x0, #1
        ret     x3
+
+unlock_skip:
+       bl      spin_unlock
+
+skip_crash_console:
+       mov     x0, #0
+       ret     x3
+
 #endif
 endfunc plat_crash_console_init
```

jwerner-chromium, 2018-11-21:
> Yes, that's likely to be the case. As I mentioned earlier, spin_locks are not expected to work with cache disabled. Basically crash reporting has to be skipped in these code paths to use the spin_lock.

Okay, checking SCTLR_C in there sounds like a good approach. But I don't think skipping crash reporting completely when the cache is disabled will be good enough for our case, unfortunately. Some of our platforms do a lot of stuff in the final part of the system suspend callback, we need to be able to get crash reports from there.

How about checking SCTLR_C and just continuing blindly if it is not set? This way, mutual exclusion is at least guaranteed as long as the caches are enabled. The chance of having two CPUs crash at the same time should already be pretty low to begin with, and then the chance that one of those happens to be in a cache-disabled path is yet smaller. I think it might not be a big deal if crash reports were broken in that very rare case, but it is important that the whole PSCI suspend/resume code path is covered.

soby-mathew, 2018-11-21:
Yes, OK. I am fine with that approach.

antonio-nino-diaz-arm, 2018-11-21:
By the way, @jwerner-chromium, we have been thinking that it would be good if you added yourself as maintainer of this driver, seeing that you are quite interested in anything going on there. What do you think?

jwerner-chromium, 2018-11-21:
Okay, I'll try to fix up my PR and upload it some time next week.

> By the way, @jwerner-chromium, we have been thinking that it would be good if you added yourself as maintainer of this driver, seeing that you are quite interested in anything going on there. What do you think?

Yes, sure, that makes sense. I can't promise any time commitment, but I'm happy to help review and maintain as my time allows.

Unfortunately I haven't found a good way to get notifications for specific files/directories on GitHub yet, or I'd be monitoring this already. I'm looking forward to when we move to Gerrit where the notification mechanisms are much nicer.

antonio-nino-diaz-arm, 2018-11-22:
You can add yourself to the maintainers.rst file. We have a script that monitors all PRs and notifies the affected maintainers. At the end of the day, it is a driver that we also use so we will also modify it and maintain it.

soby-mathew, 2018-12-12:
fixed by https://github.com/ARM-software/arm-trusted-firmware/pull/1700

closed 2018-12-12 by soby-mathew