ARM-software/tf-issues #648: Questions: wfi in EL3(PSCI CPU suspend or standby)

raghuncstate, 2018-11-15:
Hello,

I was looking at some psci code on arm platforms and was wondering if you could point me to the part of the ARM ARM that clarifies the follow behavior.
in plat/arm/css/common/css_pm.c, line 218(ATF2.0 release), i see that in the CPU standby path, the SCR_EL3.IRQ and SCR_EL3.FIQ bit are set, to enable non-secure interrupts to wake up the CPU after enter WFI. I understand the IRQ bit is for gicv2 systems in the code mentioned above so we can ignore that bit. 
However, on GICv3 systems, does setting SCR_EL3.FIQ bit to 0, and then entering WFI, preclude the CPU from ever waking up from a WFI?(assuming there are no other impdef ways of waking up WFI).
Is setting these SCR_EL3 bits a requirement per the architecture, to wake up a CPU from WFI ? if so can you point to the appropriate place in the ARM ARM ? I looked at ARM ARM table D-15, but the EA,IRQ and FIQ bits are combined in a column, so i dont know if i should interpret that as any one of the bits need to be set or if all of the bits need to be set.
To summarize: on GICv3 systems, where group0 and group1-s/ns interrupts are signaled as FIQ in EL3, and that does not have any impdef mechanism of waking up from WFI, do we require the setting of SCR_EL3.FIQ bit to 1, before entering WFI? If SCR_EL3.FIQ is 0 before entering WFI what is the expected behavior ?

soby-mathew, 2018-11-26:
Apologise for the delay as I had to check with Architecture team regarding this. 

> I understand the IRQ bit is for gicv2 systems in the code mentioned above so we can ignore that bit.

IRQ is non-secure interrupt for GICv2 systems. So if you require non-secure interrupts to wakeup from WFI on GICv2 systems.

> However, on GICv3 systems, does setting SCR_EL3.FIQ bit to 0, and then entering WFI, preclude the CPU from ever waking up from a WFI?

SCR_EL3.FIQ is for Group1 S/NS interrupts. So for these interrupts to be architecturally defined wakeup events at EL3,  SCR_EL3.FIQ bit must be set.

> Is setting these SCR_EL3 bits a requirement per the architecture, to wake up a CPU from WFI ? if so can you point to the appropriate place in the ARM ARM ? I looked at ARM ARM table D-15, but the [...]

You have to follow the trail of crumbs in the Arm Arm:
1.	From the WFI instruction you go…
2.	…and read the Wait For Interrupt section (D1.17.2 in the revision C.a) which says that the PE can stay in low power state until is receives a “WFI wake-up event”…
3.	This is later in the same section – the first bullet says that a wake-up event is a physical SError/FIQ/IRQ received by PE *that is marked A or B in table D1-1905* (Asynchronous exception masking)…
4.	… that table is in section D1.14.2. The important column is the far right (executing in EL3) and entries marked ‘C’ are “interrupt is not taken when asserted” – so we need to put the PE in the state identified by the second and final rows (only ones marked ‘B’). These require that SCR_EL3.EA, FIQ and IRQ are set to 1 – if we want the PE to wake up on all those sources.

> To summarize: on GICv3 systems, where group0 and group1-s/ns interrupts are signaled as FIQ in EL3, and that does not have any impdef mechanism of waking up from WFI, do we require the setting of [...]

Yes, SCR_EL3.FIQ has to be set as discussed above.

raghuncstate, 2018-11-26:
@soby-mathew : Thanks. Quick follow up question. Can a CPU in WFI cause EA's architecturally or through impdef mechanisms?
Also with GICv3, why would we require setting of IRQ bit when in EL3 ? As far as i'm aware, there is no way to trigger an IRQ in AARCH64 EL3 with GICv3. Is that not accurate ?

soby-mathew, 2018-11-26:
> Can a CPU in WFI cause EA's architecturally or through impdef mechanisms?

SError is defined architecturally as a valid WFI wakeup event. How a CPU delivers it would be impdef I suppose.

> Also with GICv3, why would we require setting of IRQ bit when in EL3 ? As far as i'm aware, there is no way to trigger an IRQ in AARCH64 EL3 with GICv3. Is that not accurate ?

GICv3 Group 0 interrupts trigger on IRQ line at EL3. So, if these interrupts need to a act as a wakeup source, the IRQ bit in SCR_EL3 needs to be set.

raghuncstate, 2018-11-26:
> SError is defined architecturally as a valid WFI wakeup event. How a CPU delivers it would be impdef I suppose.

@soby-mathew. Thanks for your quick response. Agree that an impdef SError during WFI might be possible.
 
> GICv3 Group 0 interrupts trigger on IRQ line at EL3. So, if these interrupts need to a act as a wakeup source, the IRQ bit in SCR_EL3 needs to be set.

Are we sure about this statement? . See table 6 in https://static.docs.arm.com/dai0492/b/GICv3_Software_Overview_Official_Release_B.pdf. Either the document is incorrect or the statement here is incorrect.
Per table 6, in AARCH64 EL3, ALL interrupts show up as FIQ. This is my understanding from the GICv3 Architecture manual as well.

soby-mathew, 2018-11-27:
> Are we sure about this statement? . See table 6 in https://static.docs.arm.com/dai0492/b/GICv3_Software_Overview_Official_Release_B.pdf. Either the document is incorrect or the statement here is incorrect.

Ah, yes, you are right. EL3 interrupts (GICv3 Group 0) always trigger on the FIQ line irrespective of the current exception level. So, yes, setting the IRQ bit in SCR_EL3 for a GICv3 based system is unnecessary.

soby-mathew, 2019-01-16:
Closing the issue.

closed 2019-01-16 by soby-mathew