ARM-software/tf-issues #650: rk3399 fails to build: region PMUSRAM overflowed

vagrantc, 2018-11-20:
Working on packaging arm-trusted-firmware in Debian.

Building arm-trusted-firmware from git commit cabe0a31801e99e7abb84d2114ded6bb56f3c71e, rk3399 fails to build with a recent Debian toolchain (gcc 8.2.0, binutils 3.31.1):

    Must remake target 'build/rk3399/debug/bl31/bl31.elf'.
  LD      build/rk3399/debug/bl31/bl31.elf
aarch64-linux-gnu-ld.bfd: build/rk3399/debug/bl31/bl31.elf section `.pmusram' will not fit in region `PMUSRAM'                         
aarch64-linux-gnu-ld.bfd: region `PMUSRAM' overflowed by 3928 bytes
make[2]: *** [Makefile:722: build/rk3399/debug/bl31/bl31.elf] Error 1                                                                  
make[2]: Leaving directory '/<<PKGBUILDDIR>>'

Full build log available at:

https://people.debian.org/~vagrant/atf/

Works fine building with the older toolchains (gcc 6, binutils 2.28) in Debian, but it's not possible to include arm-trusted-firmware in an older release; the newer toolchains are required to work for new uploads.

Thanks!

antonio-nino-diaz-arm, 2018-11-20:
@jwerner-chromium

antonio-nino-diaz-arm, 2018-11-20:
However, "overflowed by 3928 bytes" looks like there is some problem with the new version of the linker, not like GCC has become that worse in 2 major versions. There were only ~4KB of code for each CPU type (M0 and A72/53) to begin with.

jwerner-chromium, 2018-11-20:
Agreed, this looks like a bug or serious regression, not just a natural change in code generation.

@rkchrome, can Rockchip people look into this?

antonio-nino-diaz-arm, 2018-11-26:
It looks like @rkchrome is an old account and the correct one is @rockchip-linux.

antonio-nino-diaz-arm, 2019-01-28:
Which version of the compilers are you using? I'm using the following compilers from here: https://developer.arm.com/open-source/gnu-toolchain/gnu-a/downloads
```
gcc-arm-8.2-2019.01-x86_64-arm-eabi
gcc-arm-8.2-2019.01-x86_64-aarch64-linux-gnu
```
And it seems to build correctly.

maschinensehen, 2022-04-02:
I am getting a similar error on a current archlinux system with

aarch64-linux-gnu-gcc (GCC) 11.2.0
aarch64-linux-gnu-ld GNU ld (GNU Binutils) 2.38

LD      /data/arm-trusted-firmware/build/rk3399/release/bl31/bl31.elf
aarch64-linux-gnu-ld.bfd: /data/arm-trusted-firmware/build/rk3399/release/bl31/bl31.elf section `.pmusram' will not fit in region `PMUSRAM'
aarch64-linux-gnu-ld.bfd: .incbin_sram has exceeded its limit
aarch64-linux-gnu-ld.bfd: region `PMUSRAM' overflowed by 68376 bytes
make: *** [Makefile:1278: /data/arm-trusted-firmware/build/rk3399/release/bl31/bl31.elf] Error 1