ARM-software/tf-issues #652: DTB made compulsory for CoT?

Abhishek-brcm, 2018-11-27:
CoT(drivers/auth/tbbr/tbbr_cot.c) seems to be **forcing** config(dts) data to be present in key certificate.
[See tb_fw_config_hash_buf, hw_config_hash_buf, soc_fw_config_hash_buf, tos_fw_config_hash_buf, nt_fw_config_hash_buf is under authenticated_data structure].
In my case, as config data is not present in key certificate, following error comes while booting, followed by assert():
"E: Extension for data index 1 not found"

I added dummy DTS files for confirming above theory and observed that this resolves the issue.
I think we should add NEED_FDT flag in CoT to avoid error on platforms not utilizing DTS. Please advice.

soby-mathew, 2018-11-27:
@Abhishek-brcm, you are right. The `config` files are meant to be optional and the platform doesn't need to use them if not required to.

One solution is to define another CoT but without the config data so that platforms not interested in them can use that CoT. The current file can be renamed to `tbb_cot_with_config`.

Abhishek-brcm, 2018-11-28:
@soby-mathew , Thanks for the clarification.
I thought we could use NEED_FDT flag in CoT, but I realize that it would be generic across config files(tb-fw, hw, tos, nt, etc.) and would not be customize-able according to config file(s) used by given platform. In that case, it's just better to have CoT for a platform.

soby-mathew, 2018-12-04:
@Abhishek-brcm , I got around to finally testing more on this. The intention was to make the config files optional and if the platform does not have them, the default CoT as defined currently in tbbr_cot.c should work. 

So my previous statement suggesting that the CoT need to be changed when configs are absent was wrong. Apologies for the confusion.

If the config is absent, I initially assumed that the `authenticated parameter` for that config would fail (see [here](https://github.com/ARM-software/arm-trusted-firmware/blob/master/drivers/auth/auth_mod.c#L396)). But if the `authenticated parameter` is marked as `optional` then the certtool creates the authenticated parameter with dummy values (see [here](https://github.com/ARM-software/arm-trusted-firmware/blob/6c5e196be37b4afe3403f349d5da1747e12460ac/tools/cert_create/src/main.c#L462)). Since the `authenticated parameter` for the config files are all optional, it will not fail authentication if the files are not present.

Hence if you are creating the certificates using the certtool provided in the TF-A repo, you should not get an error when the config files are absent.

The platforms can create another copy of the standard CoT to save memory if not using config files. This is because the default CoT allocates memory to hold the `authenticated parameters`  for these configs.

Abhishek-brcm, 2018-12-05:
@soby-mathew , I will check on this. Thanks for digging this out.

Abhishek-brcm, 2019-01-02:
@soby-mathew , confirming that no issue when using cert-create provided in the TF-A repo.

soby-mathew, 2019-01-16:
Closing the issue.

closed 2019-01-16 by soby-mathew