ARM-software/tf-issues #653: ENABLE_PLAT_COMPAT flag are disappear on ATF v2.0

windsp, 2018-11-28:
Hi Sirs,
ENABLE_PLAT_COMPAT flag and plat/compat folder are removed in ATF 2.0 code base because of deprecated interfaces. However, we implemented some PSCI related features relying on ENABLE_PLAT_COMPAT and plat/compat mechanism. For reducing the porting effort from ATF v1.2 to ATF 2.0, may I know if we can add the ENABLE_PLAT_COMPAT flag and plat/compat folder back to local ATF v2.0 code base? Or is there any suggestion from TF team?

antonio-nino-diaz-arm, 2018-11-28:
You can rebase your patches to v1.6, update all your code to the new interfaces and then rebase on v2.0. v1.6 still has all compatibility interfaces.

sandrine-bailleux-arm, 2019-05-23:
There's been no activity in this ticket since Antonio's answer in November last year so I am closing it.

closed 2019-05-23 by sandrine-bailleux-arm