ARM-software/tf-issues #654: PSCI - CPU_SUSPEND - Standby mode

belladmv, 2018-11-29:
Hi,

We are working on SoC with Two Cortex-A35 cores.  We have enabled PSCI for CPU power management and are in the process of getting the CPU idle framework up on this SoC with Linux.

As part of this activity we have enabled the CPU_SUSPEND with only cpu standby state, 

Case 1:
We observe that one of the cores goes into WFI in EL3 as part of cpu_standby platform handler implementation (This is similar to FVP cpu_standby implementation) and the other core goes to WFI in EL1 (cpu_do_idle function in kernel)

Issue:
We observe that both the cores are in WFI (one in EL3 and the other in EL1) and are not coming out at all with any tigger such as UART. 

Case 2:
We added took reference of css_cpu_standby implementation in css_pm.c  which enables SCR_EL3.IRQ and SCR_EL3.FIQ routing to EL3. With this, we observe that core which is in WFI in EL3 comes out and functions normally by responding to UART interrupt.

Summary:-
1) With few experiments,  we found out that the core which is in WFI in EL1 is not at all coming out of WFI state. In case 2 also, the core which is in WFI in EL3 comes out of WFI state and the UART responds.

2) Without entering cpu standby in EL3, if both cores go to WFI in EL1 (arm default idle state on Linux) then both the cores are responding to interrupts and things are working fine.

Can you please let us know if we are missing something here?

Thanks,
Mahantesh/Siva.

soby-mathew, 2018-12-04:
The issue seems very strange to me. I am assuming that you are using a kernel without any modifications to the idle framework. CPUx entering EL3 and doing a WFI shouldn't affect CPUy in EL1. The only case I can think of is that the GICv3 configuration being corrupted , such that interrupts are not pending for CPUy. If there is an interrupt pending at the GIC CPU interface of CPUy, it should have woken up.

sivasanath, 2018-12-07:
Thanks Mathew for the reply, yes we have not done any modification for the cpuidle framework on kernel.
We are going through GICD,GICR and cpu interface GIC registers and kernel side code, so far they seem to be fine.

soby-mathew, 2018-12-10:
So did you verify that there is a pending interrupt at the GIC CPU interface for the CPU `not responding to WFI` at EL1 ?

sivasanath, 2018-12-11:
Here is the GIC registers dump on the CPU interface side on core-0, during EL1 WFI, where ICC_HPPIR1_EL1 shows that there is a pending interrupt 0x1E, which is 30 , which looks like arch_timer(also provide cat/proc/interrupts below).

Also some more information on the issue is, cores are getting stuck during boot, not even prompt stage.

GIC	74 of 74 registers		
ICC_AP0R0_EL1	0x00000000	32	R/W
ICC_AP0R1_EL1	Unavailable	32	R/W
ICC_AP0R2_EL1	Unavailable	32	R/W
ICC_AP0R3_EL1	Unavailable	32	R/W
ICC_AP1R0_EL1	0x00000000	32	R/W
ICC_AP1R1_EL1	Unavailable	32	R/W
ICC_AP1R2_EL1	Unavailable	32	R/W
ICC_AP1R3_EL1	Unavailable	32	R/W
S_ICC_AP1R0_EL1	0x00000000	32	R/W
S_ICC_AP1R1_EL1	Unavailable	32	R/W
S_ICC_AP1R2_EL1	Unavailable	32	R/W
S_ICC_AP1R3_EL1	Unavailable	32	R/W
N_ICC_AP1R0_EL1	0x00000000	32	R/W
N_ICC_AP1R1_EL1	Unavailable	32	R/W
N_ICC_AP1R2_EL1	Unavailable	32	R/W
N_ICC_AP1R3_EL1	Unavailable	32	R/W
ICC_ASGI1R_EL1	write only	64	WO
ICC_BPR0_EL1	0x00000002	32	R/W
ICC_BPR1_EL1	0x00000003	32	R/W
S_ICC_BPR1_EL1	0x00000002	32	R/W
N_ICC_BPR1_EL1	0x00000003	32	R/W
ICC_CTLR_EL1	0x00000402	32	R/W
S_ICC_CTLR_EL1	0x00000400	32	R/W
N_ICC_CTLR_EL1	0x00000402	32	R/W
ICC_CTLR_EL3	0x00000410	32	R/W
ICC_DIR_EL1	write only	32	WO
ICC_EOIR0_EL1	write only	32	WO
ICC_EOIR1_EL1	write only	32	WO
ICC_HPPIR0_EL1	0x000003FD	32	RO
ICC_HPPIR1_EL1	0x0000001E	32	RO
ICC_IGRPEN0_EL1	0x00000001	32	R/W
ICC_IGRPEN1_EL1	0x00000001	32	R/W
S_ICC_IGRPEN1_EL1	0x00000001	32	R/W
N_ICC_IGRPEN1_EL1	0x00000001	32	R/W
ICC_IGRPEN1_EL3	0x00000003	32	R/W
ICC_PMR_EL1	0x000000F0	32	R/W
ICC_RPR_EL1	0x000000FF	32	RO
ICC_SGI0R_EL1	write only	64	WO
ICC_SGI1R_EL1	write only	64	WO
ICC_SRE_EL1	0x00000007	32	R/W
S_ICC_SRE_EL1	0x00000007	32	R/W
N_ICC_SRE_EL1	0x00000007	32	R/W
ICC_SRE_EL2	0x0000000F	32	R/W
ICC_SRE_EL3	0x0000000F	32	R/W
ICH_AP0R0_EL2	0x00000000	32	R/W
ICH_AP0R1_EL2	Unavailable	32	R/W
ICH_AP0R2_EL2	Unavailable	32	R/W
ICH_AP0R3_EL2	Unavailable	32	R/W
ICH_AP1R0_EL2	0x00000000	32	R/W
ICH_AP1R1_EL2	Unavailable	32	R/W
ICH_AP1R2_EL2	Unavailable	32	R/W
ICH_AP1R3_EL2	Unavailable	32	R/W
ICH_EISR_EL2	0x00000000	32	RO
ICH_ELRSR_EL2	0x0000000F	32	RO
ICH_HCR_EL2	0x00000000	32	R/W
ICH_LR0_EL2	0x0000000000000000	64	R/W
ICH_LR1_EL2	0x0000000000000000	64	R/W
ICH_LR2_EL2	0x0000000000000000	64	R/W
ICH_LR3_EL2	0x0000000000000000	64	R/W
ICH_LR4_EL2	Unavailable	64	R/W
ICH_LR5_EL2	Unavailable	64	R/W
ICH_LR6_EL2	Unavailable	64	R/W
ICH_LR7_EL2	Unavailable	64	R/W
ICH_LR8_EL2	Unavailable	64	R/W
ICH_LR9_EL2	Unavailable	64	R/W
ICH_LR10_EL2	Unavailable	64	R/W
ICH_LR11_EL2	Unavailable	64	R/W
ICH_LR12_EL2	Unavailable	64	R/W
ICH_LR13_EL2	Unavailable	64	R/W
ICH_LR14_EL2	Unavailable	64	R/W
ICH_LR15_EL2	Unavailable	64	R/W
ICH_MISR_EL2	0x00000000	32	RO
ICH_VMCR_EL2	0x004C0008	32	R/W
ICH_VTR_EL2	0x90080003	32	RO

**************************************************************************************************
/ # cat /proc/interrupts
           CPU0       CPU1
  2:       5363       2778     GICv3  30 Level     arch_timer
  5:        343          0     GICv3  32 Level     c000000.uart
  6:          0          0     GICv3  53 Level     a105000.pdma
  7:          0          0     GICv3  88 Level     a105000.pdma
  8:          0          0     GICv3  89 Level     a105000.pdma
  9:          0          0     GICv3  90 Level     a105000.pdma
 10:          0          0     GICv3  91 Level     a105000.pdma
 11:          0          0     GICv3  92 Level     a105000.pdma
 12:          0          0     GICv3  93 Level     a105000.pdma
 13:          0          0     GICv3  94 Level     a105000.pdma
 14:          0          0     GICv3  95 Level     a105000.pdma
 15:      14869          0     GICv3  36 Level     c004000.spi
 16:          0          0     GICv3  37 Level     c005000.spi
 17:         99          0     GICv3  49 Level     mmc0
 18:          0          0     GICv3  47 Level     mmc1
I

soby-mathew, 2019-06-07:
@sivasanath , Is this issue resolved ?

sivasanath, 2019-06-10:
@soby-mathew , sorry, we are no more looking in to this as we moved out to a different project, but issue was not resolved, thanks for the support

soby-mathew, 2019-06-10:
OK, I will close the issue for now. We can reopen it later when you are ready to pursue it.

closed 2019-06-10 by soby-mathew