ARM-software/tf-issues #655: Trustedfirmware.org Project Migration Information

joannafarley-arm, 2018-12-05:
On October 16th it was announced that the Arm Trusted Firmware project for A class processors currently hosted here on Github was becoming part of the  Trusted Firmware (https://www.trustedfirmware.org) project which is a Linaro Community Projects Division open project. There is a Blog post (https://www.trustedfirmware.org/news/linaro-community-projects-division-announces-the-trusted-firmware-open-project/) that provides more details of the announcement.
 
At this time Github provides the repository along with a Wiki and Issues list. It is the intent to migrate all these to the Trusted Firmware (https://www.trustedfirmware.org) project site using the gerrit, wiki and tasks/issues systems provided there. The expected timeline for this migration is by mid 1Q2019 and once this is completed the Github project will become read only with all requests redirected to go to the TrustedFirmware.org site.
 
As part of the migration the project copyright guidance is being updated to reflect the new open status. Project licensing has not changed.
 
As further details of the migration of facilities and processes to use are finalised this Github Issue will be updated with that information. In the longer term future additional services hope to be offered by the Trusted Firmware project site such as a build system, a CI and automated test system and test board farm.
 
Shortly after the publication of the aforementioned blog post, the first public release of the Trusted Firmware-A Tests was announced on November 8th. The following blog post provides more details: https://www.trustedfirmware.org/blog/trusted-firmware-a-tests-release/
 
TF-A Tests is a suite of bare-metal tests to exercise the Trusted Firmware-A features from the Normal world. It enables strong TF-A functional testing without dependency on a Rich OS, mainly interacting with TF-A through its SMC interface. More details can be found on the project's landing page:
https://git.trustedfirmware.org/TF-A/tf-a-tests.git/about/
 
Mailing lists are now available for both Trusted Firmware-A and Trusted Firmware-A Tests: https://www.trustedfirmware.org/contact/
 
Many thanks
Joanna Farley

antonio-nino-diaz-arm, 2018-12-05:
The copyright guidelines have been updated in https://github.com/ARM-software/arm-trusted-firmware/pull/1709

mcarlini, 2018-12-17:
A clarification on the above change: the contribution guidelines now allow contributors to add their own company/individual name into copyright notices for newly introduced files and for significant contributions to existing files.

However, the “_Arm Limited and Contributors_” copyright notice is retained on existing files, along with the associated acknowledgements.rst file, to cover legacy contributors to the project and avoid churn in the code.

Ultimately, the copyright notices are only an indication of the file's copyright status; the actual status is determined by the commit history.