ARM-software/tf-issues #656: Renesas rcar-gen3: cert_header_sa6 Compilation Issue

AlexeiFedorov, 2018-12-12:
tools\renesas\rcar_layout_create\sa6.c has

#include <stdint.h>

which causes compilation error on Yocto:

| aarch64-poky-linux-gcc -DRCAR_SA0_SIZE=1 -DRCAR_SA6_TYPE=0 -I../../include/lib/stdlib -D=AARCH64  -c -o sa6.o sa6.c
| In file included from sa6.c:7:0:
| /home/alexei/Work/Yocto_3.13_20181205/build/tmp/work/h3ulcb-poky-linux/arm-tf/2.0+gitAUTOINC+19b56cf4a2_556d7d9e3b-r0/recipe-sysroot-native/usr/lib/aarch64-poky-linux/gcc/aarch64-poky-linux/7.3.0/include/stdint.h:9:16: fatal error: stdint.h: No such file or directory
|  # include_next <stdint.h>
|                 ^~~~~~~~~~
| compilation terminated.
| <builtin>: recipe for target 'sa6.o' failed
| make[1]: *** [sa6.o] Error 1
| plat/renesas/rcar/platform.mk:403: recipe for target 'rcar_layout_tool' failed
| make: *** [rcar_layout_tool] Error 2
| ERROR: oe_runmake failed

There's no compilation error with gcc-arm-8.2-2018.08-i686-mingw32-aarch64-elf tool chain, because compiler
searches for <stdint.h> in
\gcc-arm-8.2-2018.08-i686-mingw32-aarch64-elf\lib\gcc\aarch64-elf\8.2.1\include
finding "stdint.h" with
# include_next <stdint.h>
and then getting "stdint.h" from

gcc-arm-8.2-2018.08-i686-mingw32-aarch64-elf\aarch64-elf\include folder.

Trusted Firmware-A Porting Guide reads that
"To avoid subtle toolchain behavioral dependencies, the header files provided
by the compiler are not used. The software is built with the ``-nostdinc`` flag
to ensure no headers are included from the toolchain inadvertently. ",

tools\renesas\rcar_layout_create\makefile has flags defined as:
CFLAGS += ${DEFINES}
CFLAGS += -I../../include/lib/stdlib

and adding "-nostdinc" flag will make gcc-arm-8.2-2018.08-i686-mingw32-aarch64-elf compilation also  fail because

../../include/lib/stdlib is a wrong path (taken from Renesas ARM-TF version 1.5 "tools\dummy_create\makefile") to

include\lib\libc\stdint.h for version ARM-TF version 2.0.

This issue can be fixed by changing CFLAGS to

CFLAGS += -nostdinc \
                    -I../../../include/lib/libc \
                    -I../../../include/lib/libc/aarch64

antonio-nino-diaz-arm, 2018-12-12:
> To avoid subtle toolchain behavioral dependencies, the header files provided
> by the compiler are not used. The software is built with the -nostdinc flag
> to ensure no headers are included from the toolchain inadvertently.

This only applies to the firmware itself, not to the tools. The tools are free to use the system's libc.

antonio-nino-diaz-arm, 2018-12-12:
@ldts

AlexeiFedorov, 2018-12-12:
The source file sa6.c which is causing compilation error is used to create cert_header_sa6.bin  and prodice cert_header_sa6.srec file which is then flashed to the board. It's not a tool but a part of firmware.

antonio-nino-diaz-arm, 2018-12-12:
Ok, I see. If you already have a patch to fix it you could raise a pull request.