ARM-software/tf-issues #657: march32-directive for clang?

Yann-lms, 2018-12-13:
When I tried to compile TF-A for STM32MP1 with clang, I have several errors.
One is about thumb flags and is corrected with https://github.com/ARM-software/arm-trusted-firmware/pull/1719

But the other is this one:
`In file included from plat/st/stm32mp1/bl2_io_storage.c:7:`
`include/lib/aarch32/arch_helpers.h:191:1: error: invalid instruction`
`DEFINE_SYSOP_TYPE_FUNC(dmb, sy)`
`^`
`include/lib/aarch32/arch_helpers.h:167:11: note: expanded from macro` `'DEFINE_SYSOP_TYPE_FUNC'`
`        __asm__ (#_op " " #_type);                                      \`
`                 ^`
`<scratch space>:8:1: note: expanded from here`
`"dmb"`
`^`
`<inline asm>:1:2: note: instantiated into assembly here`
`        dmb sy`
`        ^`
`1 error generated.`

One solution for me is to change in Makefile:
`-TF_CFLAGS_aarch32	=	$(target32-directive)`
`+TF_CFLAGS_aarch32	=	$(target32-directive) $(march32-directive)`

But I don't know if it is the correct one.

dp-arm, 2018-12-17:
The way we are handling this currently is a bit messy and will probably require revisiting at some point.  I think your suggestion is probably fine for the moment.  Can you raise a PR with the fix?

Yann-lms, 2018-12-18:
Sure, I will push the patch.

closed 2018-12-19 by antonio-nino-diaz-arm