ARM-software/tf-issues #658: Error with raspberry pi 3 using optee - mmap_add_region_check() failed. error -22

mgallissot, 2018-12-20:
Hello everyone,

I've having an error while updating the ATF and using it with optee on a raspberry pi 3.
I'm compiling the ATF with the following command 👍 
```
make   PLAT=rpi3 DEBUG=1   CROSS_COMPILE=aarch64-linux-gnu- \
   NEED_BL32=yes  BL32=../optee_os/optee_os/out/arm/core/tee-header_v2.bin \
   BL32_EXTRA1=../optee_os/optee_os/out/arm/core/tee-pager_v2.bin  
   BL32_EXTRA2=../optee_os/optee_os/out/arm/core/tee-pageable_v2.bin \
   MBEDTLS_DIR=../mbedtls-2.13.0 BL33=../u-boot/u-boot.bin CRASH_REPORTING=1 \
   GENERATE_COT=1   SAVE_KEYS=1   TRUSTED_WORLD_KEY=./keys/trusted_world.key   
   NON_TRUSTED_WORLD_KEY=./keys/nt_worlded.key  \
   ROT_KEY=./keys/rot.key   SCP_BL2_KEY=./keys/scp_bl2_content.key   \
   BL31_KEY=./keys/soc_content.key   BL32_KEY=./keys/tos_content.key   
   BL33_KEY=./keys/nt_fw_content.key   SPD=opteed   all fip
```
At boot, I've got the following error when BL31 boot
```
ERROR: mmap_add_region_check() failed. error -22
ASSERT: lib/xlat_tables_v2/xlat_tables_core.c:471
```

antonio-nino-diaz-arm, 2018-12-20:
Which commit of the TF are you using?

mgallissot, 2018-12-20:
I'm using latest one (commit c8765826f4c2d10db0b660defccc84f7bce11af0)

mgallissot, 2018-12-20:
Thanks Antonio,

To be sure I've cleaned my local folder with 
```
git reset --hard HEAD
git clean -xffd
git pull
```
Here is the last screen output with the higher log level. 
```
NOTICE:  BL2: v2.0(debug):v2.0-431-gc8765826
NOTICE:  BL2: Built : 10:46:38, Dec 20 2018
INFO:    BL2: Doing platform setup
INFO:    BL2: Loading image id 3
VERBOSE: Using Memmap
VERBOSE: FIP header looks OK.
VERBOSE: Using FIP
INFO:    Loading image id=3 at address 0x100e0000
INFO:    Image id=3 loaded: 0x100e0000 - 0x100eb078
INFO:    BL2: Loading image id 4
VERBOSE: Using Memmap
VERBOSE: FIP header looks OK.
VERBOSE: Using FIP
INFO:    Loading image id=4 at address 0x10100000
INFO:    Image id=4 loaded: 0x10100000 - 0x1010001c
INFO:    OPTEE ep=0x10100000
INFO:    OPTEE header info:
INFO:          magic=0x4554504f
INFO:          version=0x2
INFO:          arch=0x1
INFO:          flags=0x0
INFO:          nb_images=0x1
INFO:    BL2: Loading image id 21
VERBOSE: Using Memmap
VERBOSE: FIP header looks OK.
VERBOSE: Using FIP
INFO:    Loading image id=21 at address 0x10100000
INFO:    Image id=21 loaded: 0x10100000 - 0x10146500
INFO:    BL2: Skip loading image id 22
INFO:    BL2: Loading image id 5
VERBOSE: Using Memmap
VERBOSE: FIP header looks OK.
VERBOSE: Using FIP
INFO:    Loading image id=5 at address 0x11000000
INFO:    Image id=5 loaded: 0x11000000 - 0x1109ef98
NOTICE:  BL1: Booting BL31
INFO:    Entry point address = 0x100e0000
INFO:    SPSR = 0x3cd
VERBOSE: Argument #0 = 0x100bf490
VERBOSE: Argument #1 = 0xf1e2d3c4b5a6978
VERBOSE: Argument #2 = 0x0
VERBOSE: Argument #3 = 0x0
VERBOSE: Argument #4 = 0x0
VERBOSE: Argument #5 = 0x0
VERBOSE: Argument #6 = 0x0
VERBOSE: Argument #7 = 0x0
VERBOSE: Trusted SRAM seen by this BL image: 0x100e0000 - 0x100f6000
VERBOSE: Code region: 0x100e0000 - 0x100e8000
VERBOSE: Read-only data region: 0x100e8000 - 0x100eb000
VERBOSE: Coherent region: 0x100f5000 - 0x100f6000
ERROR:   mmap_add_region_check() failed. error -22
ASSERT: lib/xlat_tables_v2/xlat_tables_core.c:741:false
BACKTRACE: START: assert
0: EL3: 0x100e46fc
1: EL3: 0x100e6c8c
2: EL3: 0x100e53f0
3: EL3: 0x100e54f8
4: EL3: 0x100e4988
5: EL3: 0x100e0358
6: EL3: 0x100e00c0
7: EL3: 0x100b4090
BACKTRACE: END: assert
```

antonio-nino-diaz-arm, 2018-12-20:
Ok, this is still not enough information.

Go to ``mmap_add_region_check()`` and add this like to the start:

printf("0x%llx 0x%lx 0x%lx 0x%lx\n", mm->base_pa, mm->base_va, mm->siz  e, mm->granularity)

antonio-nino-diaz-arm, 2018-12-20:
There is your problem:

```
0x1 0x1 0x10000 0x40000000
```

For some reason the platform is trying to map a region at address 0x1, which isn't valid because it's not aligned to 4KB. This region is most likely the DTB one:

https://github.com/ARM-software/arm-trusted-firmware/blob/c8765826f4c2d10db0b660defccc84f7bce11af0/plat/rpi3/rpi3_common.c#L83

I suspect that ``RPI3_PRELOADED_DTB_BASE`` is being defined to 1 for some reason. You need to build the TF with this option set to the address of the DTB (check your config.txt to know the right address).

mgallissot, 2018-12-21:
Thanks, I've forced the RPI3_PRELOADED_DTB_BASE value and it's fine now.

closed 2018-12-21 by mgallissot