ARM-software/tf-issues #660: Could compile error alert if HW_ASSISTED_COHERENCY is misconfigured with cortex-a55/a75 built-in?

leon-chen-mtk, 2018-12-25:
Hi,
We notice that there is no L1/L2 cache maintenance in CPU/cluster core power down operations in CORTEX A55, A75 cortex_a55.S/cortex_a75.S if HW_ASSISTED_COHERENCY is disabled.
So it implies that if system uses cortex a55/a75, developer must enable HW_ASSISTED_COHERENCY specifically.
However, if we miss this option enabled while using cortext a55/a75, system behaves weirdly if cpu wakes up from suspend(e.g, data abort randomly since stale data resides in external memory while dcache is not enabled early w/o HW_ASSISTED_COHERENCY).
It is not easy to figure out the root cause in first scene.

If there is a compile error indication when developer build in cortex a55/a75 without HW_ASSISTED_COHERENCY enabled, that will be good for project development. How do you think?

soby-mathew, 2018-12-26:
Hi @leon-chen-mtk ,
That is a good suggestion. Yes, the CPUs you have mentioned should always be built with HW_ASSISTED_COHERENCY build flag enabled. Could you please submit a PR to that effect ? In fact all the CPUs which do not have cache maintenance in their power down sequence should have the flag enabled,

[Most of the team are away on holiday and hence our response may be delayed]

soby-mathew, 2019-01-08:
@leon-chen-mtk, will you be able to submit a PR to check the build flag for these CPUs ?

leon-chen-mtk, 2019-01-09:
@soby-mathew, I think there is easy way or extensible way.
Easy way is just check HW_ASSISTED_COHERENCY defined or not in cortxt_a55.h, cortex_a75.h
Extensible way maybe have a arch_compile dependency file, list CPUs and which feature use(e.g, DynamIQ), and MUST option associated CPUs.(so far, HW_ASSISTED_COHERENCY), makefile retrieve BL31_SOURCES to match arch_compile listed cpu name and MUST option. Not sure is this useful.

soby-mathew, 2019-01-16:
@leon-chen-mtk, I think the easy way should suffice for now.

antonio-nino-diaz-arm, 2019-04-24:
https://review.trustedfirmware.org/#/c/TF-A/trusted-firmware-a/+/927/

jts-arm, 2019-05-03:
Is fixed by the following patch: https://review.trustedfirmware.org/#/c/TF-A/trusted-firmware-a/+/927/

closed 2019-05-03 by jts-arm