ARM-software/tf-issues #662: BL31: Add support for scatter files

vwadekar, 2019-01-05:
NVIDIA plans to use the TF-A codebase with the 'armlink' linker instead of GCC. 'armlink' expects a new linker script called as a scatter file as it does not fully support ld.S scripts.

This issue will track the work required to introduce a generic scatter file for the BL31 code, that can be (re)used by all platforms.

These are the tentative sub-tasks
- Converge all linker symbols to bl_common.h as new macros
- Update code to use these newly introduced macros
- Update the makefiles to use a different command when 'armlink' is detected
- Update documentation with relevant information to use 'armlink' with BL31
- Update any Tegra code, if required

**Tentative ETA:** 1/18

**Note**: the plan is to use local Tegra builds as much as possible to verify these changes and use the ARM build system to compile for all other platforms.

soby-mathew, 2019-01-08:
@vwadekar, we had done a preliminary evaluation for using the scatter gather format within the team and found the following limitations (but you may have found solutions for them) : 

1. Support for dynamic relocation/Position independent support executable (PIE) for AArch64 does not seem to be present.
2. Overlay capability is insufficient to support existing FVP build.
3. Forward references of linker symbols are not allowed in SG format. We use this capability for specifying overlays from within platform linker script which references symbols defined by later by main linker script.
4. Ability to specify `exclude` file list from sections is lacking.

Hence I think it might be difficult to support this format generically. But please raise a PR and we can evaluate it.

vwadekar, 2019-01-15:
Partial changes: https://github.com/ARM-software/arm-trusted-firmware/pull/1759

vwadekar, 2019-01-15:
The actual scatter file is part of PR https://github.com/ARM-software/arm-trusted-firmware/pull/1751

closed 2019-06-07 by soby-mathew