ARM-software/tf-issues #663: Rockchip RK3328 GICv2 broken.

vicencb, 2019-01-06:
When testing ATF v2.0 and also 8e7b27a488acb1dbc8bf407626e7fc2f700223bd there is an assertion failure appearing on the console on a RK3328 platform:
```
INFO: ARM GICv2 driver initialized
ASSERT: drivers/arm/gic/v2/gicv2_helpers.c:193
```
the assertion code is `assert(prop_desc->intr_grp == GICV2_INTR_GROUP0);` which ensures that the interrupt is from group 0.

On the file `plat/rockchip/common/rockchip_gicv2.c:25` there is a comment stating that `On a GICv2 system, the Group 1 secure interrupts are treated as Group 0 interrupts.` just before declaring an array of interrups from group 1 as group 0.

Both files are clearly in conflict and this is the reason for the assertion failure.

This issue is not present in v1.6.

antonio-nino-diaz-arm, 2019-01-08:
The problem was probably introduced in https://github.com/ARM-software/arm-trusted-firmware/commit/2d6f1f01b141775a192e36d120073f67b7b378c2

I guess this is restricted to RK3328, right? We've had some contributions affecting rockchip since then, and this would have been noticed fairly quickly.

soby-mathew, 2019-01-08:
>#define PLAT_RK_GICV2_G1S_IRQS						\
	INTR_PROP_DESC(RK_IRQ_SEC_PHY_TIMER, GIC_HIGHEST_SEC_PRIORITY,	\
		       GICV2_INTR_GROUP1, GIC_INTR_CFG_LEVEL),		\
	INTR_PROP_DESC(RK_IRQ_SEC_SGI_6, GIC_HIGHEST_SEC_PRIORITY,	\
		       GICV2_INTR_GROUP1, GIC_INTR_CFG_LEVEL)

Group1 interrupts are non-secure for GICv2. It seems trying to unify the definitions for GICv3 and GICv2 seems to have caused the problem . If the above macro is corrected, the problem may be solved,

antonio-nino-diaz-arm, 2019-01-28:
@vicencb, could you try this branch and see if it works for you? https://github.com/antonio-nino-diaz-arm/arm-trusted-firmware/tree/an/rk-gic

antonio-nino-diaz-arm, 2019-01-28:
Ok, I will create a PR. Thank @soby-mathew for the fix, by the way, I only wrote the code!

closed 2019-01-31 by antonio-nino-diaz-arm