ARM-software/tf-issues #664: Remove recursion from xlat_tables_v2 library

vwadekar, 2019-01-11:
The xlat_tables_v2 library contains recursion cases which are not allowed under the current MISRA rules.

These are the functions that recursively call themselves:

1. xlat_tables_unmap_region
2. xlat_tables_map_region

The code needs to be reorganized to fix these problems

antonio-nino-diaz-arm, 2019-01-15:
I guess you are talking about rule 17.2. According to the MISRA C 2012 document, the rationale for this is that using recursion has the danger of using more stack than the one that is actually available unless recursion is very tightly controlled.

The function that uses the biggest amount of stack ```xlat_tables_map_region()```. In this case we know that the worst case is 3 levels of recursion. Also, this level of recursion is always reached during boot because the text/rodata/bss/etc sections are 4KB, which forces the mapping code to go to the deepest level of recursion. This means that if there is not enough stack for the code, the platform will simply not boot at all.

I do agree that it would be nice to not have recursion there. For starters, it could make the code a bit more readable. However, this is not a priority as the benefits of doing it don't justify the amount of effort it needs.

soby-mathew, 2019-01-16:
Hi @vwadekar,
In this case, the worst case recursion is known and is tightly controlled. MISRA gives you the room to not comply with a rule as long as proper justification can be provided. 

But I agree, it is good if the recursion can be changed to an iterative logic. Since you say this is something important for you, please come up with a patch for the change and we can integrate it after review.

DavidPu, 2019-02-27:
Hello, I raised PR to convert recursion to iteration. this PR depends on PR for #673

closed 2019-03-13 by soby-mathew