ARM-software/tf-issues #666: Question about arguments passing in SMCCC

leon-chen-mtk, 2019-01-16:
Hi,
I am confused about arguments passing in SMC Calling Convention.
The spec. mentions that x0~x6 are taken as arguments(64 bit as an example).
However, SMC handler C function prototype only receives x0~x4(x0 as SMC function ID), x5~x6 registers are omitted.

So caller can only pass x0~x4 even caller assigns x0~x7 from EL1 or EL2.
Does ARM TF not comply with SMCCC spec. or which part I misunderstand?

antonio-nino-diaz-arm, 2019-01-16:
You can access the rest of the registers like this:
https://github.com/ARM-software/arm-trusted-firmware/blob/fe77b53e32692fc5d35df208e2b3d21a28f6228b/services/std_svc/spm/spci.c#L712-L714
Is this what you wanted?

leon-chen-mtk, 2019-01-16:
Oh, got it, thank you a lot!

closed 2019-01-16 by leon-chen-mtk