ARM-software/tf-issues #669: Source code verification with sparse

laroche, 2019-01-27:
Would you be open to check ATF with the sparse source code checker
that is available via https://sparse.wiki.kernel.org/index.php/Main_Page
or accept patches fixing its findings?

One of the biggest findings is, that you do not define var/function declarations
in a few files. I assume most of this might be not including the proper *.h files.

Example: lib/libc/memchr.c:9:6: warning: symbol 'memchr' was not declared. Should it be static?
Just including include/lib/libc/string.h fixes this.

best regards,

Florian La Roche

soby-mathew, 2019-01-28:
Yes, looking at the [man page](https://linux.die.net/man/1/sparse), the  check looks useful in improving code quality and finding issues. We would be happy to receive contributions fixing up problems detected.

laroche, 2019-01-28:
I'll prepare a few lines on how to invoke sparse on ATF source and propose a few fixes.
Overall this should not result in huge changes.

Coming the next few days...,

Florian La Roche

jwerner-chromium, 2019-01-31:
FWIW, you don't need sparse to find that sort of issue (although it can probably find other things). You just need to add -Wmissing-prototypes to CFLAGS. Most other projects set that (e.g. coreboot, Linux with W=1), it would probably be a good idea to have it in TF as well.

closed 2019-06-07 by soby-mathew