ARM-software/tf-issues #670: Decouple platforms from plat/arm

antonio-nino-diaz-arm, 2019-01-28:
There are some non-Arm platforms that are using files inside the plat/arm tree.

- mt6795, mt8173: @mtk09422
- zynqmp: @sivadur
- allwinner: @Andre-ARM, @smaeul.
- qemu: @jenswi-linaro

It would be helpful to break the dependencies. If there are some functions/macros that can be helpful outside of Arm platforms, it would be a good idea to mention it. We have already done it in the past: https://github.com/ARM-software/arm-trusted-firmware/pull/1647/commits/0916c38deca4a2e26720da6b1aa1e15de87f9a34

sandeepbrcm, 2019-03-15:
From a first look I think plat_arm_gic_driver_xxx  arm_gicv3.c , arm_gicv2.c  can be moved out of plat/arm ?  The wrappers are just to choose between v2 and v3.

antonio-nino-diaz-arm, 2019-04-03:
- ``plat/arm/common/arm_gicv2.c`` and ``plat/arm/common/arm_gicv3.c``: I agree that most likely they can be shared, but right now this might be difficult. They are weak functions, which makes me think that one wrong move would break platforms silently. Also, they are all using "arm platforms" names, so all Arm platforms would have to be changed.
- ``plat_arm_gic_driver_xxx``: Sorry, I don't know which file you're talking about, could you tell us the full path?

sandeepbrcm, 2019-04-03:
Thanks.  plat_arm_gic_driver_xxx by this I meant all the generic functions in arm gicv2 gicv3 drivers, nothing specific. I agree that all arm platforms would have to be changed. I am planing to upstream a Broadcom SoC support. I am going to take a replica of these files into plat/brcm initially followed by an attempt to move gicv2 gicv3 to common.

antonio-nino-diaz-arm, 2019-04-04:
Yes, I think that in this case the best option is to copy the files and change the "plat_arm" prefixes to "brcm" or similar. This is not ideal, though, because other platforms don't get to benefit from changes we do to said files. However, that also means that we won't break other platforms by mistake while modifying Arm platforms.

sandeepbrcm, 2019-04-04:
Thanks. One more question on this is: - should arm spdx header be retained in those files, where probably only changes are the prefixes ?

antonio-nino-diaz-arm, 2019-04-05:
The copyright line? I guess so, and update the year.

sandeepbrcm, 2019-04-06:
> The copyright line? I guess so, and update the year.

Thanks.

closed 2019-06-07 by soby-mathew