ARM-software/tf-issues #673: Cyclomatic Complexity of less than 10

vwadekar, 2019-02-04:
Cyclomatic complexity is well described in posts [1] and [2]. NVIDIA is interested in keeping this number less than 10, as part of its safety effort, to reduce the number of decisions and associated testing effort.

We have prepared some changes downstream to fix the issues flagged by Coverity. We would like to push the relevant changes upstream. Is this something that would be beneficial to others?

[1] https://blogs.msdn.microsoft.com/zainnab/2011/05/17/code-metrics-cyclomatic-complexity/
[2] https://en.wikipedia.org/wiki/Cyclomatic_complexity

soby-mathew, 2019-02-08:
Reducing complexity within the code base is something we would be interested in. Please raise the PR and we can review the changes.

vwadekar, 2019-02-08:
@soby-mathew 

Sounds good.

DavidPu, 2019-02-27:
Hello, I created a PR to try to reduce CCM for xlat_tables_map_region and xlat_tables_unmap_region functions.

vwadekar, 2019-03-01:
I think there are some other changes required before this issue can be closed. The source of confusion might be the "fixes" comment in David's PR.

@DavidPu can you push the remaining changes as well?

closed 2019-03-02 by rkchrome