ARM-software/tf-issues #674: Trusty fails to build with generic services on Nvidia platforms

antonio-nino-diaz-arm, 2019-02-12:
Nvidia platforms with ``SPD=trusty`` and ``TRUSTY_SPD_WITH_GENERIC_SERVICES=1`` fail to build:

```
make PLAT=tegra TARGET_SOC=t210 SPD=trusty \
TRUSTY_SPD_WITH_GENERIC_SERVICES=1
```

This happens because the code of Trusty expects the defines ``PLAT_ARM_GICC_BASE`` and ``PLAT_ARM_GICD_BASE`` to be present, but they only exist in Arm platforms, where this code builds correctly. Nvidia platforms define ``TEGRA_GICD_BASE`` and  ``TEGRA_GICC_BASE``.

This issue isn't about fixing this, though. We would like to know why this is needed at all. If it isn't needed, we think that the best solution would be to remove the interfaces completely.

CC: @vwadekar

vwadekar, 2019-02-12:
@antonio-nino-diaz-arm 

Thanks for flagging this issue. Tegra platforms do not support `generic-arm64-smcall.c` so you can safely remove this build configuration.

AFAIR, someone from Google added that code as they wanted to run Trusty with QEMU

antonio-nino-diaz-arm, 2019-02-13:
So I guess they only used it to prototype something, the current QEMU port is missing the definitions that would be needed for this.

We can keep building it for FVP indefinitely in case it breaks, but I think that if this isn't useful it should be removed, as we are not testing it and it seems that nobody else is either.

vwadekar, 2019-02-13:
@antonio-nino-diaz-arm 

I can only speak for the Tegra platforms. As of 2/13/19 (9am PST), we do not plan to use `TRUSTY_SPD_WITH_GENERIC_SERVICES=1`.

I think we should mark it as deprecated and keep building it with FVP for the time being. You can try to ask "Arve Hjønnevåg <arve@android.com>" on the side if he has any plans to keep supporting this feature for QEMU before removing it.

antonio-nino-diaz-arm, 2019-02-15:
Thanks, @vwadekar. @arve-android, do you have anything to add to this?

soby-mathew, 2019-06-07:
The query seems to be answered. Closing the issue.

closed 2019-06-07 by soby-mathew