ARM-software/tf-issues #675: compile error

chenbd, 2019-02-23:
The master branch compile error using the following command:
`make PLAT=fvp RESET_TO_BL31=1 ARM_LINUX_KERNEL_AS_BL33=1 PRELOADED_BL33_BASE=0xc0000000 ARM_PRELOADED_DTB_BASE=0x83e00000`

error messages:
PP bl1/bl1.ld.S
LD build/fvp/release/bl1/bl1.elf
aarch64-linux-gnu-ld.bfd: build/fvp/release/bl1/bl1.elf: Not enough room for program headers, try linking with -N
aarch64-linux-gnu-ld.bfd: final link failed: Bad value

But v1.5 tag compiles OK.

antonio-nino-diaz-arm, 2019-02-25:
Can you try adding ``ENABLE_PIE=0``?

What version of GCC are you using?

chenbd, 2019-02-26:
Hello,

I'm using `gcc-linaro-7.2.1-2017.11-x86_64_aarch64-linux-gnu`.

add `ENABLE_PIE=0` fixed this issue.
Thanks.

antonio-nino-diaz-arm, 2019-02-26:
@soby-mathew

soby-mathew, 2019-02-26:
From the looks of it, seems to be a linker issue.  Either the `-pie` option passed to linker or object files compiled with `-fpie` seems to be a problem.

closed 2019-03-13 by soby-mathew