ARM-software/tf-issues #676: chain-of-trust descriptor optimisations

JoelHutton, 2019-02-26:
The chain-of-trust description was using a lot of memory, as it was a sparse struct array. The image 
descriptors contained in this array also contained arrays, which were not always used.

The first patch in the series replaces the chain of trust descriptor array with a struct pointer array,
https://gerrit.oss.arm.com/#/c/126822/6

The second patch changes the struct declaration (and adds casts in the assignment) to replace the arrays with pointers, allowing the uninitialised arrays to default to NULL.
https://gerrit.oss.arm.com/#/c/126494/10

Altogether, these patches save the following memory from bl1 and bl2 respectively, a more detailed breakdown can be seen in the commit message.
bl1:-7427B, bl2:-2660B

soby-mathew, 2019-02-26:
Just to note, the TF-A defines a default CoT Trust [here](https://github.com/ARM-software/arm-trusted-firmware/blob/master/drivers/auth/tbbr/tbbr_cot.c#L111). The platforms are allowed to override the default CoT and implement their own variant. This would break for platforms which have their own CoT and they would need to migrate to the new structure. This should be fairly easy to do and would have significant memory savings.

soby-mathew, 2019-02-26:
The patch-set is currently undergoing review. Please let us know if there are any comments for this change.

soby-mathew, 2019-02-26:
Edited the description to point to trustedfirmware.org gerrit links.

sandrine-bailleux-arm, 2019-05-23:
Both patches ([here](https://review.trustedfirmware.org/c/TF-A/trusted-firmware-a/+/657) and [here](https://review.trustedfirmware.org/c/TF-A/trusted-firmware-a/+/658)) have now been merged.

closed 2019-05-23 by sandrine-bailleux-arm