ARM-software/tf-issues #677: Dependency between TRUSTED_BOARD_BOOT and mbedtls

ruchi393, 2019-02-28:
Hi,

I recently see a patch which adds the following in the  file plat/common/plat_bl_common.c.

#if TRUSTED_BOARD_BOOT
#include <drivers/auth/mbedtls/mbedtls_config.h>
#endif

On my platform , I have a usecase where TRUSTED_BOARD_BOOT is required but I am not compiling mbedtls. This addition in the plat_bl_common.c file breaks this use case.

Can this dependency be removed from this file ?

Regards,
Ruchika

antonio-nino-diaz-arm, 2019-02-28:
@jts-arm

jts-arm, 2019-02-28:
Hello @ruchi393,
Could you please copy-paste the compilation error you are facing?

Thank you!
John

ruchi393, 2019-02-28:
CC      plat/common/plat_bl_common.c
In file included from plat/common/plat_bl_common.c:14:0:
include/drivers/auth/mbedtls/mbedtls_config.h:92:10: fatal error: mbedtls/check_config.h: No such file or directory
 #include "mbedtls/check_config.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.

jts-arm, 2019-02-28:
Could you please try the following? From the plat_bl_common.c file:
1) remove the mbedtls_config.h inclusion
2) remove the plat_get_mbedtls_heap() function.

Try this quick fix, please, to see if compilation succeeds.

ruchi393, 2019-03-01:
I had tried removing the mbedtls_config.h inclusion which helps in resolving compilation issue since 
plat_get_mbedtls_heap doesn't get used in my code. 

Probably this function needs to move to some other file.

antonio-nino-diaz-arm, 2019-03-01:
Can you try with https://github.com/ARM-software/arm-trusted-firmware/pull/1847?

antonio-nino-diaz-arm, 2019-03-05:
Fixed in https://github.com/ARM-software/arm-trusted-firmware/pull/1847

closed 2019-03-05 by antonio-nino-diaz-arm