ARM-software/tf-issues #685: query on license header

sandeepbrcm, 2019-03-14:
I have a generic query on BSD-3. Please advise even if its not much TF-A specific but
there are code from plat/arm which fits with/without modification to some platforms.
some have replicated and rename files/functions and some platform may have a different or slightly resembling implementation.

In order to add a platform support the code reused from plat/arm reference should
have which header if there are considerable additions/modifications to the original file/functions.
Only:
  Copyright (c) 2018, ARM Limited and Contributors. All rights reserved
Or
  Copyright (c) 2018,  <OWNER> . All rights reserved
Or  can it also have both.
  Copyright (c) 2018, ARM Limited and Contributors. All rights reserved
  Copyright (c) 2018,  <OWNER> . All rights reserved
I see most plat/xxx  implementation maintains header as
  Copyright (c) 2018, ARM Limited and Contributors. All rights reserved
Or it depends on what exactly the file looks like. 
Thanks.

robertovargas-arm, 2019-04-02:
After the 2.o release the policy about copyrights was changed. Before 2.0, all the files had "ARM Limited and Contributors. All rights reserved", but after it, significant modifications can incorporate Copyrights from companies different than ARM. Lines like:

      Copyright (c) 2018, . All rights reserved

seem wrong to me, because the copyright holder is lost, and it means that the copyright line has no meaning.

sandeepbrcm, 2019-04-02:
Thanks . I omitted the name. I meant   Copyright (c) 2018, contributor-name.  All rights reserved.
But I understood that part. I understand that in case of significant contribution one can
add own copyright.  I am specifically confused with the files that are inspired / replicated from
plat/arm to plat/xxx  eg:  arm_bl2_setup.c.

Or in some cases is it even valid to have ..
Copyright (c) 2018, ARM Limited and Contributors. All rights reserved
Copyright (c) 2018, other-contributor-name. All rights reserved

soby-mathew, 2019-06-07:
Yes, it is valid to have multiple copyright licenses in a file.

closed 2019-06-07 by soby-mathew