ARM-software/tf-issues #689: CCN: Is there any API to flush the cache content beyond PoC(i.e., L3 Cache) to reflect the latest copy on DRAM.

pangupta, 2019-03-26:
Is there any API to flush the cache content beyond PoC(i.e., L3 Cache in case of CCN) to reflect the latest copy on DRAM.

For SoC using CCN as interconnect, there are scenarios where the content on L3(PoC) cache never reaches to DRAM leading to stale copy of data on DRAM. This further leads to hang or crash issues.

Recommended solution provided by ARM is to use the non-architectural ways to flush L3 cache.
That is by using Power Management operation on L3 cache to flush its data to DRAM.

I am ready with the API to be part of CCN to flush the L3 cache using Power management operation.

Please share your view.

soby-mathew, 2019-03-28:
No, currently there is no API to flush L3 by Setway.  This is a useful API to introduce. Please raise a PR for the same.

soby-mathew, 2019-06-07:
@pangupta, Please send patch to trustedfirmware.org. Closing the issue.

closed 2019-06-07 by soby-mathew