ARM-software/tf-issues #691: Trusted Firmware version 2.1 is now available

joannafarley-arm, 2019-03-30:
Trusted Firmware version 2.1 is now available and can be found here:
https://github.com/ARM-software/arm-trusted-firmware/tree/v2.1

Please refer to the [readme](https://github.com/ARM-software/arm-trusted-firmware/blob/master/readme.rst) and [change log](https://github.com/ARM-software/arm-trusted-firmware/blob/master/docs/change-log.rst) for further information.

Regards

Joanna

vagrantc, 2019-04-03:
I've been maintaining arm-trusted-firmware in Debian, and would like to figure out a trust path for updated versions...

I see that the v2.1 tag is gpg-signed, but with a public key not present in the gpg keyring network, so a bit hard to independently verify. Github apparently knows about the key, as it's marking it as verified, though I can't seem to find any way to get github to give me the public key.

Would it be possible to publish the public key used for signing releases somewhere?

live well,
  vagrant

joannafarley-arm, 2019-04-04:
H Vagrant,

Just wanted to let you know I’m working on a solution for you.

Regards

Joanna

joannafarley-arm, 2019-04-05:
Hi Vagrant,

The public key has been uploaded to http://keyserver.pgp.com/ and is available there and hopefully will be distributed round the gpg keyring network.

Let men know if you have any issues.

Regards

Joanna

sandrine-bailleux-arm, 2019-05-23:
@vagrantc I hope Joanna's solution works for you. Closing this ticket for now.

closed 2019-05-23 by sandrine-bailleux-arm