ARM-software/tf-issues #692: Removal of deprecated interface

ambroise-arm, 2019-04-04:
The legacy console API is going to be removed in favor of `MULTI_CONSOLE_API` as a result of the v2.1 release, as stated in the "TF-A Release Information" document and as mentioned in #613.

The following platforms are affected and need to update their interface:

- imx8m* @JackyBai 

- qemu @jenswi-linaro 

- uniphier @masahir0y 

- rcar @ldts @marex 

- zynqmp @sivadur 

- tegra @vwadekar 

- mt8173 @mtk09422 

- warp7 @bryanodonoghue @niej

sivadur, 2019-04-05:
@antonio-nino-diaz-arm   whats the plan? when are you going to remove it?
Will look into zynqmp part next week. Hope this is ok.

antonio-nino-diaz-arm, 2019-04-05:
The idea is to give you a couple of months to migrate, but we haven't really discussed when it will be removed.

sivadur, 2019-04-05:
@antonio-nino-diaz-arm  Oh ok, Thanks for the info.