ARM-software/tf-issues #693: L2 dormant mode on A53

Bigfan-Chen, 2019-04-10:
Hi ARM friends,
As far as I know, for a powerdown request in CPU_SUSPEND flow, current PSCI implementation always flushs the cache in function psci_do_pwrdown_cache_maintenance.
If we would like to implement L2 dormant mode on A53, could you share a guideline to us? And, would you add helper function to porting layer to support this?
Thanks!!

Fan

jts-arm, 2019-04-15:
Dear Fan,

Trusted Firmware-A has migrated since last week to trustedfirmware.org. Could you please raise this issue in the following page? https://developer.trustedfirmware.org/project/view/1/
You can login there just by using your GitHub account so it is very easy and straightforward! We will be happy to discuss the issue there.

Thank you!
John

soby-mathew, 2019-06-07:
Issue is discussed here : https://developer.trustedfirmware.org/Q19

closed 2019-06-07 by soby-mathew