ARM-software/tf-issues #694: What is the purpose of MULTI_CONSOLE_API?

vabbarap, 2019-04-10:


jwerner-chromium, 2019-04-11:
The main purpose is to be able to have more than one console driver registered at the same time. For example, a platform could both register a UART driver to output logs on the serial console and a memory console driver to output logs into a memory buffer where they can be retrieved by the operating system.

A secondary purpose was to allow merging normal and crash console output for consoles that support this. With the legacy console API, every platform had to manually implement plat_crash_console_init which made it very hard to use non-hardcoded console parameters (e.g. if the console is selected by runtime configuration). With the multi-console API, platforms can link plat/common/crash_console_helpers.S which will automatically use the console drivers that are already registered for normal console output in case of a crash.

vabbarap, 2019-04-12:
Thanks for the info.

closed 2019-05-23 by sandrine-bailleux-arm