ARM-software/tf-issues #695: Upgrading PSCI from v1.0 to v1.1

MarekBykowski, 2019-04-10:
Hi,

I'm doing the work of upgrading PSCI to v1.1. 

It is clear for me from that the v1.1 introduces SYSTEM_RESET2, MEM_PROTECT (and MEM_PROTECT_CHECK_RANGE) and I have to port/implement them. However I have trouble determining what the following delta is (referred to from http://infocenter.arm.com/help/topic/com.arm.doc.den0022d/Power_State_Coordination_Interface_PDD_v1_1_DEN0022D.pdf):

- Added additional details on registers that should be initialized when EL2 is not used
- Updated save restore language to make it more generic. See section6.8.
- Removed big.LITTLE migration language
- Added flexibility in encoding of power state parameters in statisticscalls
- Added clarification to GICv3 ITS save and restore

I cannot find any specific commits integrating the points above on "integration" branch. Should I just go ahead and not really bother with it.

Thanks,
Marek

soby-mathew, 2019-04-12:
This is more clarification in the specification and many of them were already catered for in the implementation. I don't think you should worry about them.

sandrine-bailleux-arm, 2019-05-23:
I believe Soby has answered the question so I am closing this ticket. If you feel you need more details, could you please take the discussion on the [TF-A mailing list](https://lists.trustedfirmware.org/mailman/listinfo/tf-a)? Thank you.

closed 2019-05-23 by sandrine-bailleux-arm