ARM-software/tf-issues #696: Unable to rebuild the ATF when we add any macro in the make command

vabbarap, 2019-04-12:
First time say we build the ATF using the below command
make PLAT=zynqmp RESET_TO_BL31=1
It builds successfully.

Again now we want to add another macro added like this
make PLAT=zynqmp RESET_TO_BL31=1 ZYNQMP_CONSOLE=cadence1

It wont rebuild again.

Does anyone have faced this issue?

JoelHutton, 2019-04-12:
This is a current limitation of our build system. As we use make, build options changing does not prompt a recompilation of the affected files. This leads to unexpected results when building twice with different build options.
If you need to build twice, you should use 'make clean' or 'make realclean' in between to remove build artefacts.

vabbarap, 2019-04-12:
Thanks for the info.
Any plan to resolve this in future releases?

soby-mathew, 2019-04-15:
The build flag dependencies are not propagated correctly in the current build system. There is a desire to overhaul the build system to be config based. But this is not scheduled yet.

sandrine-bailleux-arm, 2019-05-23:
I believe Soby has answered the question. Thus, closing this ticket.

closed 2019-05-23 by sandrine-bailleux-arm