ARM-software/tf-issues #698: Error while using udelay function on xilinx zynqmp

vabbarap, 2019-10-22:
Getting error in the udelay function
assert(usec < UINT32_MAX / timer_ops->clk_div);

In the function 
void generic_delay_timer_init(void) {
 unsigned int div  = plat_get_syscnt_freq2(); 
Here the value of div frequency is 99.99MHz(99990008) which is cauisng the error.

Is there any way to get rid of this error which has frequencies other than 100MHz, 200MHZ and so on.

ASSERT: drivers/delay_timer/delay_timer.c:30
BACKTRACE: START: assert
0: EL3: 0x6b80
1: EL3: 0x90f8
2: EL3: 0x7454
3: EL3: 0x3970
4: EL3: 0x20e4
5: EL3: 0x87ec
BACKTRACE: END: assert

Thanks
Venkatesh

max-shvetsov, 2019-10-31:
Greetings @vabbarap.
I have made a fix for your issue, please have a look at [this patch](https://review.trustedfirmware.org/c/TF-A/trusted-firmware-a/+/2403). I did my best to replicate the issue by tweaking _fvp_ platform as we don't have your specific platform. Please let us know if you have issues on your platform.
Also please use [issue tracking board](https://developer.trustedfirmware.org/maniphest/query/open/) in the future.
Kind regards,
Max

vabbarap, 2019-11-05:
@max-shvetsov 
Thanks for the patch.It worked.

Thanks
Venkatesh

closed 2019-11-05 by vabbarap