ARM-software/tf-issues #699: Git tag for 2.2 release has not been applied to GitHub mirror

pbatard, 2019-11-02:
Because of the popularity of the platform, many people are looking for TF-A releases in the GitHub mirror at https://git.trustedfirmware.org/TF-A/trusted-firmware-a.git/.

However, even though v2.2 was released 11 days ago as per https://git.trustedfirmware.org/TF-A/trusted-firmware-a.git/, the GitHub repo still hasn't been updated to create a new v2.2 tag as well as a new GitHub release (https://github.com/ARM-software/arm-trusted-firmware/releases).

Can you please make sure that tags and new releases get applied to GitHub on release day? If not, I feel that a lot of people, as was the case for myself until I decided to look at git.trustedfirmware.org, will be very confused as to why there exist what looks like release commits for v2.2 on GitHub, yet no official tag for it or release artefacts.

Thank you.

joannafarley-arm, 2019-11-02:
Hi Pete, thanks for your message. 

The primary repository for TF-A since may 2019 is now https://git.trustedfirmware.org/TF-A/trusted-firmware-a.git/?h=master and we have attempted to communicate that people should refer there. However, we have some automated scripts to keep a GitHub readonly mirror as a convenience and as you point out the v2.2 tags are not listed on GitHub. Thanks for bringing that to our attention. We will see what we can do to address that but people should look towards https://git.trustedfirmware.org/TF-A/trusted-firmware-a.git/?h=master as the primary repository and subscribe to the project mail alias https://lists.trustedfirmware.org/mailman/listinfo/tf-a for news and updates.

Thanks

Joanna

joannafarley-arm, 2019-11-05:
Issue addressed by an update to the syncing scripts.

closed 2019-11-05 by joannafarley-arm