ARM-software/tf-issues #709: stm32mp157c-dk2 error in configuring TRUSTED_BOARD_BOOT

zhengtf0963, 2021-03-30:
I am trying to configure the arm trusted firmware of the stm board, but when I place TRUSTED_BOARD_BOOT=1 \GENERATE_COT=1 \, the compilation will fail, and the error code is as follows:
`
/home/tflgr/stm32_test/build/../toolchains/aarch32/bin/arm-linux-gnueabihf-ld.bfd: ./build/stm32mp1/debug/bl2/bl2_main.o: in function `bl2_main':
/home/tflgr/stm32_test/trusted-firmware-a/bl2/bl2_main.c:90: undefined reference to `auth_mod_init'
/home/tflgr/stm32_test/build/../toolchains/aarch32/bin/arm-linux-gnueabihf-ld.bfd: ./build/stm32mp1/debug/bl2/bl_common.o: in function `load_auth_image_recursive':
/home/tflgr/stm32_test/trusted-firmware-a/common/bl_common.c:178: undefined reference to `auth_mod_get_parent_id'
/home/tflgr/stm32_test/build/../toolchains/aarch32/bin/arm-linux-gnueabihf-ld.bfd: /home/tflgr/stm32_test/trusted-firmware-a/common/bl_common.c:193: undefined reference to `auth_mod_verify_img'
Makefile:1053: recipe for target 'build/stm32mp1/debug/bl2/bl2.elf' failed
make[1]: *** [build/stm32mp1/debug/bl2/bl2.elf] Error 1

`

Yann-lms, 2021-03-31:
Hello @zhengtf0963,

Several features are not supported in upstream TF-A for STM32MP1 platform.
You should check STMicroelectronics github repo:
https://github.com/STMicroelectronics/arm-trusted-firmware

Best regards,
Yann

Yann-lms, 2024-08-16:
The support of Trusted Boot for STM32MP1 has been merged in this series: https://review.trustedfirmware.org/q/topic:%22stm32mp1-trusted-boot%22
And is integrated since TF-A v2.8.

closed 2024-08-16 by Yann-lms