ARM-software/tf-issues #710: RESET_TO_BL31: accepting params

ldts, 2022-03-30:
Would something like this be acceptable.

Author: Jorge Ramirez-Ortiz <jorge@foundries.io>
Date:   29 minutes ago

    bl31: aarch64: RESET_TO_BL31 with parameters
    
    It is not always the case that RESET_TO_BL31 enabled platforms don't execute a
    bootloader before BL31. For those use cases, being able to receive arguments
    from that first loader (i.e: a DTB) might be necessary feature.
    
    This code has been validated on iMX8mm.
    
    Signed-off-by: Jorge Ramirez-Ortiz <jorge@foundries.io>

diff --git a/bl31/aarch64/bl31_entrypoint.S b/bl31/aarch64/bl31_entrypoint.S
index 2d672dd12..4e492bad4 100644
--- a/bl31/aarch64/bl31_entrypoint.S
+++ b/bl31/aarch64/bl31_entrypoint.S
@@ -67,6 +67,7 @@ func bl31_entrypoint
                _exception_vectors=runtime_exceptions           \
                _pie_fixup_size=BL31_LIMIT - BL31_BASE
 
+#ifndef RESET_TO_BL31_WITH_PARAMS
        /* ---------------------------------------------------------------------
         * For RESET_TO_BL31 systems, BL31 is the first bootloader to run so
         * there's no argument to relay from a previous bootloader. Zero the
@@ -77,6 +78,7 @@ func bl31_entrypoint
        mov     x21, 0
        mov     x22, 0
        mov     x23, 0
+#endif
 #endif /* RESET_TO_BL31 */
 
        /* --------------------------------------------------------------------

ldts, 2022-03-30:
It has been a while since I contributed to this project and the process is not quite clear to me (not sure where to send my patch despite having read the docs TBH). thanks,

sandrine-bailleux-arm, 2022-04-01:
Hi @ldts ,

> It has been a while since I contributed to this project and the 'new' process is not quite clear to me (not sure where to send my patch despite having read the docs TBH).

Please send your patches to this Gerrit server: https://review.trustedfirmware.org

If you need help setting up your environment to contribute patches there, there is some documentation hosted [here](https://tf-ci-users-guide.readthedocs.io/en/latest/#how-to-contribute-code). Is that the one you went through already? Do you have any specific questions, specific steps you're blocked on or error messages you're getting?

Regards,
Sandrine

ldts, 2022-04-01:
hi @sandrine-bailleux-arm thanks for coming back to me - yes I had forgotten about setting up gerrit. it is all good now I think

https://review.trustedfirmware.org/c/TF-A/trusted-firmware-a/+/14596 

I'll close this issue now

closed 2022-04-01 by ldts