ARM-software/tf-issues #715: mvebu marvell espressobin v7 atf-marvell 18.12 directory mv_ddr was not found

house-chang, 2024-07-09:
mvebu marvell espressobin v7 atf-marvell 18.12 (昇-1.5-armada-18.12)
directory : ./drivers/marvell/mv_ddr/ was not found, where do I find ? thank you.

code and readme are in https://github.com/MarvellEmbeddedProcessors/atf-marvell/tree/atf-v1.5-armada-18.12

readme : atf-marvell/docs/marvell/build.txt :
(2) Set path for MV_DDR component::

		> export MV_DDR_PATH=path/to/mv_ddr

	For A7/8K, use this parameter to point to mv_ddr driver sources to allow BLE build.
	For A37x0, it is used for ddr_tool build.

	Usage example: MV_DDR_PATH=path/to/mv_ddr

	The parameter is optional for A7/8K, when this parameter is not set, the mv_ddr
	sources are expected to be located at: drivers/marvell/mv_ddr. However, the parameter
	is necessary for A37x0.

Yann-lms, 2024-07-09:
Hi @house-chang,

Usually binaries are not in the official TrustredFirmware-A repository. OEMs have their own repositories to store that kind of binaries.
If you look at this page:
https://trustedfirmware-a.readthedocs.io/en/latest/plat/marvell/armada/build.html
you may find the repository:
https://github.com/MarvellEmbeddedProcessors/mv-ddr-marvell.git

But I'm not familiar with Marvell platforms, maybe @kostapr could confirm?

kostapr, 2024-07-09:
Guys, I am not with Marvell for more than a year now. I expect that somebody was assigned by company to handle the platform related issues in mainline sources. Sorry, but I cannot help you with it...